openwisp/openwisp-controller #18: Add CA segfaults

peterhassing, 2017-11-23:
Hi,

On trying to save a new CA I get a "502 Bad Gateway".

Running version: VERSION = (0, 2, 4, 'final') (version 0, 2, 3, 'final' does not seem to have this problem)

this is the trace:

[pid: 2541|app: 0|req: 8/36] 192.168.1.124 () {48 vars in 983 bytes} [Thu Nov 23 13:21:34 2017] GET /admin/pki/ca/add/ => generated 15791 bytes in 88 msecs (HTTP/1.1 200) 7 headers in 382 bytes (1 switches on core 1)
!!! uWSGI process 3478 got Segmentation Fault !!!
*** backtrace of 3478 ***
/opt/openwisp2/env/bin/uwsgi(uwsgi_backtrace+0x2c) [0x46ae9c]
/opt/openwisp2/env/bin/uwsgi(uwsgi_segfault+0x21) [0x46b261]
/lib/x86_64-linux-gnu/libc.so.6(+0x354b0) [0x7fe2d78454b0]
/opt/openwisp2/env/lib/python3.5/site-packages/cryptography/hazmat/bindings/_openssl.abi3.so(+0x126600) [0x7fe2d2d7d600]
/opt/openwisp2/env/lib/python3.5/site-packages/cryptography/hazmat/bindings/_openssl.abi3.so(+0x1df288) [0x7fe2d2e36288]
/opt/openwisp2/env/lib/python3.5/site-packages/cryptography/hazmat/bindings/_openssl.abi3.so(+0x10a5a4) [0x7fe2d2d615a4]
/opt/openwisp2/env/lib/python3.5/site-packages/cryptography/hazmat/bindings/_openssl.abi3.so(+0x10aeca) [0x7fe2d2d61eca]
/opt/openwisp2/env/lib/python3.5/site-packages/cryptography/hazmat/bindings/_openssl.abi3.so(+0xfbea3) [0x7fe2d2d52ea3]
/opt/openwisp2/env/lib/python3.5/site-packages/cryptography/hazmat/bindings/_openssl.abi3.so(+0x1df691) [0x7fe2d2e36691]
/opt/openwisp2/env/lib/python3.5/site-packages/cryptography/hazmat/bindings/_openssl.abi3.so(+0x7120b) [0x7fe2d2cc820b]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyCFunction_Call+0xc9) [0x7fe2d7c88039]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x7555) [0x7fe2d7d941b5]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x79d9) [0x7fe2d7d94639]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x79d9) [0x7fe2d7d94639]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x79d9) [0x7fe2d7d94639]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0x24ac6c) [0x7fe2d7e24c6c]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x62d9) [0x7fe2d7d92f39]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x79d9) [0x7fe2d7d94639]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x79d9) [0x7fe2d7d94639]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0x24ac6c) [0x7fe2d7e24c6c]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalCodeEx+0x23) [0x7fe2d7e24d43]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0xd2bd5) [0x7fe2d7cacbd5]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyObject_Call+0x6e) [0x7fe2d7d6154e]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x2d8b) [0x7fe2d7d8f9eb]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0x24ac6c) [0x7fe2d7e24c6c]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalCodeEx+0x23) [0x7fe2d7e24d43]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0xd2bd5) [0x7fe2d7cacbd5]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyObject_Call+0x6e) [0x7fe2d7d6154e]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x2d8b) [0x7fe2d7d8f9eb]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0x24ac6c) [0x7fe2d7e24c6c]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalCodeEx+0x23) [0x7fe2d7e24d43]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0xd2bd5) [0x7fe2d7cacbd5]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyObject_Call+0x6e) [0x7fe2d7d6154e]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x2d8b) [0x7fe2d7d8f9eb]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0x24ac6c) [0x7fe2d7e24c6c]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x62d9) [0x7fe2d7d92f39]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0x24ac6c) [0x7fe2d7e24c6c]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x62d9) [0x7fe2d7d92f39]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0x24ac6c) [0x7fe2d7e24c6c]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalCodeEx+0x23) [0x7fe2d7e24d43]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0xd2bd5) [0x7fe2d7cacbd5]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyObject_Call+0x6e) [0x7fe2d7d6154e]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x2d8b) [0x7fe2d7d8f9eb]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0x24ac6c) [0x7fe2d7e24c6c]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalCodeEx+0x23) [0x7fe2d7e24d43]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0xd2bd5) [0x7fe2d7cacbd5]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyObject_Call+0x6e) [0x7fe2d7d6154e]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x2d8b) [0x7fe2d7d8f9eb]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0x24ac6c) [0x7fe2d7e24c6c]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalCodeEx+0x23) [0x7fe2d7e24d43]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0xd2bd5) [0x7fe2d7cacbd5]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyObject_Call+0x6e) [0x7fe2d7d6154e]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x2d8b) [0x7fe2d7d8f9eb]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0x24ac6c) [0x7fe2d7e24c6c]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalCodeEx+0x23) [0x7fe2d7e24d43]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0xd2bd5) [0x7fe2d7cacbd5]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyObject_Call+0x6e) [0x7fe2d7d6154e]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x2d8b) [0x7fe2d7d8f9eb]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0x24ac6c) [0x7fe2d7e24c6c]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalCodeEx+0x23) [0x7fe2d7e24d43]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(+0xd2bd5) [0x7fe2d7cacbd5]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyObject_Call+0x6e) [0x7fe2d7d6154e]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x2d8b) [0x7fe2d7d8f9eb]
/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x79d9) [0x7fe2d7d94639]
*** end of backtrace ***
DAMN ! worker 1 (pid: 3478) died :( trying respawn ...
Respawned uWSGI worker 1 (new pid: 3484)

I've tried to go through the commits between the versions but I'm kinda out of my league here...

Thanx,
Peter.

nemesisdesign, 2017-11-23:
Hi @peterhassing,

seems like https://github.com/openwisp/django-x509/issues/14, which triggered https://github.com/pyca/cryptography/issues/4015 which should be fixed but doesn't seem released yet.

As a temporary workaround, you can try downgrading the version of the python cryptography library.

BTW, are you encountering this issue on Docker? I couldn't replicate it on a VM.

peterhassing, 2017-11-23:
Hi nemesisdesign

Yeah, I thought that could be it too, but after downgrading cryptography the problem persists..

sudo pip2 list | grep crypt
asn1crypto (0.23.0)
cryptography (2.1.2)
pycrypto (2.6.1)

This is running on a native 16.04 ubuntu, no virtualisation.

nemesisdesign, 2017-11-23:
check the application log and see if you can find any stacktrace there, it's likley a different problem, are you importing a CA or creating a new one?

peterhassing, 2017-11-23:
I'm creating a new CA.

What application log are you referring to ?

nemesisdesign, 2017-11-23:
`tail -n 100 /opt/openwisp2/log/error.log` unless you chose a different directory for openwisp2

peterhassing, 2017-11-23:
/opt/openwisp2/log/error.log only has mention of a different problem (a few hours ago)
/opt/openwisp2/log/nginx.error.log is full of these 
2017/11/23 13:21:36 [error] 2057#2057: *83 upstream prematurely closed connection while reading response header from upstream, client: 192.168.1.124, server: pb.cannyware.net, request: "POST /admin/pki/ca/add/ HTTP/1.1", upstream: "uwsgi://127.0.0.1:3031", host: "pb.cannyware.net", referrer: "https://pb.cannyware.net/admin/pki/ca/add/"

peterhassing, 2017-11-23:
A lot of these in the error.log, they were solved after I added an organisation WITHOUT adding an organisation user.

[ERROR 2017-11-23 12:45:44,981] module: exception, process: 3143, thread: 140612286871296
Internal Server Error: /admin/openwisp_users/organization/e2812312-7b6c-42f9-a87b-30c8bfc1d628/change/
Traceback (most recent call last):
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/core/handlers/exception.py", line 41, in inner
    response = get_response(request)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/core/handlers/base.py", line 249, in _legacy_get_response
    response = self._get_response(request)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/core/handlers/base.py", line 187, in _get_response
    response = self.process_exception_by_middleware(e, request)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/core/handlers/base.py", line 185, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/contrib/admin/options.py", line 551, in wrapper
    return self.admin_site.admin_view(view)(*args, **kwargs)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/utils/decorators.py", line 149, in _wrapped_view
    response = view_func(request, *args, **kwargs)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/views/decorators/cache.py", line 57, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/contrib/admin/sites.py", line 224, in inner
    return view(request, *args, **kwargs)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/reversion/admin.py", line 174, in change_view
    return super(VersionAdmin, self).change_view(request, object_id, form_url, extra_context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/contrib/admin/options.py", line 1511, in change_view
    return self.changeform_view(request, object_id, form_url, extra_context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/utils/decorators.py", line 67, in _wrapper
    return bound_func(*args, **kwargs)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/utils/decorators.py", line 149, in _wrapped_view
    response = view_func(request, *args, **kwargs)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/utils/decorators.py", line 63, in bound_func
    return func.__get__(self, type(self))(*args2, **kwargs2)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/contrib/admin/options.py", line 1408, in changeform_view
    return self._changeform_view(request, object_id, form_url, extra_context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/contrib/admin/options.py", line 1449, in _changeform_view
    self.save_related(request, form, formsets, not add)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/contrib/admin/options.py", line 1003, in save_related
    self.save_formset(request, form, formset, change=change)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/contrib/admin/options.py", line 991, in save_formset
    formset.save()
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/forms/models.py", line 666, in save
    return self.save_existing_objects(commit) + self.save_new_objects(commit)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/forms/models.py", line 800, in save_new_objects
    self.new_objects.append(self.save_new(form, commit=commit))
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/forms/models.py", line 949, in save_new
    obj.save()
  File "/opt/openwisp2/env/lib/python3.5/site-packages/organizations/abstract.py", line 247, in save
    raise OrganizationMismatch
organizations.exceptions.OrganizationMismatch

nemesisdesign, 2017-11-23:
mm if you can manage to find the procedure to replicate these bugs I can prepare a proper bug report for each and we can get them fixed with the help of the [Google Code-In OpenWISP students and mentors](https://codein.withgoogle.com/organizations/openwisp/).

nemesisdesign, 2017-12-06:
@peterhassing I just released a new version of one of the dependencies ([django-x509 0.3.2](https://pypi.python.org/pypi/django-x509/0.3.2)) which I hope fixes this issue. If you could try upgrading your instance, ensure django-x509 0.3.2 is installed and try again it would be **VERY** helpful.

nemesisdesign, 2017-12-19:
This issue should be fixed by the latest release, if after upgrading and reloading application you still encounter the problem don't hesitate to report. Thank you @peterhassing.

closed 2017-12-19 by nemesisdesign