openwisp/openwisp-controller #24: Error saving Device without organization

HizkiFW, 2017-12-06:
The Web UI displays a 500 Server Error when trying to save or preview the configuration of a new device which doesn't have an Organization selected. Here's a relevant snippet on `/opt/openwisp2/log/error.log`:

```
[ERROR 2017-12-06 13:10:11,135] module: exception, process: 11785, thread: 140183607027456
Internal Server Error: /admin/config/device/add/
Traceback (most recent call last):
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/core/handlers/exception.py", line 41, in inner
    response = get_response(request)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/core/handlers/base.py", line 249, in _legacy_get_response
    response = self._get_response(request)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/core/handlers/base.py", line 187, in _get_response
    response = self.process_exception_by_middleware(e, request)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/core/handlers/base.py", line 185, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/contrib/admin/options.py", line 551, in wrapper
    return self.admin_site.admin_view(view)(*args, **kwargs)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/utils/decorators.py", line 149, in _wrapped_view
    response = view_func(request, *args, **kwargs)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/views/decorators/cache.py", line 57, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/contrib/admin/sites.py", line 224, in inner
    return view(request, *args, **kwargs)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/openwisp_controller/config/admin.py", line 63, in add_view
    return super(DeviceAdmin, self).add_view(request, form_url, extra_context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django_netjsonconfig/base/admin.py", line 77, in add_view
    return super(BaseConfigAdmin, self).add_view(request, form_url, extra_context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/reversion/admin.py", line 177, in add_view
    return super(VersionAdmin, self).add_view(request, form_url, extra_context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/contrib/admin/options.py", line 1508, in add_view
    return self.changeform_view(request, None, form_url, extra_context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/utils/decorators.py", line 67, in _wrapper
    return bound_func(*args, **kwargs)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/utils/decorators.py", line 149, in _wrapped_view
    response = view_func(request, *args, **kwargs)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/utils/decorators.py", line 63, in bound_func
    return func.__get__(self, type(self))(*args2, **kwargs2)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/contrib/admin/options.py", line 1408, in changeform_view
    return self._changeform_view(request, object_id, form_url, extra_context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/contrib/admin/options.py", line 1447, in _changeform_view
    if all_valid(formsets) and form_validated:
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/forms/formsets.py", line 461, in all_valid
    if not formset.is_valid():
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/forms/formsets.py", line 321, in is_valid
    self.errors
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/forms/formsets.py", line 295, in errors
    self.full_clean()
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/forms/formsets.py", line 349, in full_clean
    self._errors.append(form.errors)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/forms/forms.py", line 175, in errors
    self.full_clean()
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/forms/forms.py", line 386, in full_clean
    self._post_clean()
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/forms/models.py", line 413, in _post_clean
    self.instance.full_clean(exclude=exclude, validate_unique=False)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/db/models/base.py", line 1235, in full_clean
    self.clean()
  File "/opt/openwisp2/env/lib/python3.5/site-packages/openwisp_controller/config/models.py", line 100, in clean
    self.organization = self.device.organization
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/db/models/fields/related_descriptors.py", line 194, in __get__
    "%s has no %s." % (self.field.model.__name__, self.field.name)
django.db.models.fields.related_descriptors.RelatedObjectDoesNotExist: Device has no organization.
```

**Steps to reproduce**:  
1. Login to the admin panel and add a new device
2. Add a configuration, or use a template
3. Make sure the `Organization` dropdown is left at `---------`
4. Click on <kbd>Preview configuration</kbd> or any of the save buttons.

Here's a video demonstrating the bug: https://www.youtube.com/watch?v=Gp1wE9DqZ48

NoumbissiValere, 2019-01-25:
is this issue still open? i will like to solve it

NoumbissiValere, 2019-01-27:
I think a way to solve this problem will be to validate the organization field before the Device is saved. if the Device doesn't have an organization field set, a ValidateError except should be raised. or what do you think @atb00ker

atb00ker, 2019-01-29:
Yes and adding `blank=False` in form should be helpful for users! :smile: 
Feel free to work on it! :smile:

NoumbissiValere, 2019-01-29:
Yes. i was expecting organization to be be a field in the Device model but i didn't see that. please can you give me a hint on where this is defined? thanks

atb00ker, 2019-01-29:
Please checkout `openwisp-utils` module! :smile:

NoumbissiValere, 2019-01-29:
ok thanks @atb00ker

PR #85 (2019-02-08): [config/models] Fixed Error saving Device without organization #24
In the config class, device.organization was assigned to organization
without checking if the device actually had the organization or not.
Performing this check before assignment fixed the issue.

Fixes #24
commit a3025a80 (2019-02-08): [config/models] Fixed Error saving Device without organization #24

In the config class, device.organization was assigned to organization
without checking if the device actually had the organization or not.
Performing this check before assignment fixed the issue.
To test this change, an automated test was added which entails the
creation of a config_device with a device that has no organization.
a ValidationError is expected else the test fails.

Fixes #24
closed 2019-02-08 by nemesisdesign