openwisp/openwisp-controller #29: Re-registration of a device deleted with django-netjsonconfig fails

okraits, 2018-03-20:
What I did:

* let openwisp-config register the device at the openwisp controller
* delete the device on the controller with django-netjsonconfig

What I saw:

* when trying to pull configuration changes, openwisp-config logs the following message:

    `daemon.err openwisp: Failed to retrieve checksum: HTTP/1.1 404 Not Found`

* The device is not re-registered at the controller although it has both uuid and key

Is that

* intended behaviour so that a possibly stolen device can not re-register itself at the controller
* a bug?

nemesisdesign, 2018-03-20:
seems like intended behaviour, if you delete the device from the openwisp2 database, the checksum and download config operations will return 404 because won't find any device with that UUID and key, if you want the device to register again you have to restore its shared secret

okraits, 2018-03-20:
Yes, you have to restore the shared secret and unset the uuid and key, then the client is able to register the device again.

It doesn't prevent stolen devices from re-registering here either because the default config is still available in ROM.

I'm closing this issue as this is intended behaviour.

closed 2018-03-20 by okraits

nemesisdesign, 2018-03-20:
True, stolen devices can be problematic. A solution to that could be to modify the firmware image in order to have a strong password configured for the failsafe mode, do you think that's doable?

okraits, 2018-03-22:
In our images there's even no failsafe mode available :-) If you're interested I can find out how that is done.

Regarding the lock out of stolen devices, I guess we have to store the openwisp-config configuration file (and thus the shared secret) in the overlay so that it can't be restored from ROM.

Another thing to consider for such a lock-out feature would be that locked-out devices can't establish a VPN connection anymore. What do you think about that?

nemesisdesign, 2018-03-22:
The VPN lock out can be done by revoking the certificate from the OpenWISP web UI, but the openvpn server must be configured in order to preiodically download the CRL from OpenWISP and reload the openvpn config accordingly when a change is detected.