openwisp/openwisp-controller #30: [connections] Add possibility to connect to devices

nemesisdesign, 2018-05-04:
Not so long ago I've experimented with the **optional possibility of initiating connections to network devices** to launch operations, eg: update a configuration as soon as it's changed in OpenWISP.

This feature is the base that will enable us to implement features like:

- updating the configuration as soon as it's changed in OpenWISP
- allowing administrators to open SSH shell sessions from the web UI 
- performing firmware upgrades
- adding SNMP connection configurations for monitoring/management purposes
- adding support for firmwares which have HTTP interfaces/API

At the moment I will focus just on enabling the possibility to instantiate SSH connections to devices in order to perform configuration updates.

cappe87, 2018-05-04:
Hi!
thank you for the update and the works.

For SSH connections,which ip are you using? the LAST IP or ip of the vpn?

bye

nemesisdesign, 2018-05-04:
The prototype I built 8 months ago allowed users to specify different ip addresses: https://github.com/openwisp/openwisp-controller/commit/8fdf726950cc7c9204aaba784fb4a26a36468d63

I'll resume from there, clean it up and simplify.

nemesisdesign, 2018-05-06:
@cappe87 opened PR #31

nemesisdesign, 2018-09-05:
Only automated tests are missing from #31.

NoumbissiValere, 2019-02-04:
@nemesisdesign  i which to help out with this test so that the work can be merged . please can you give me an idea on how to go about it? thanks.

nemesisdesign, 2019-04-22:
#31 merged! :tada:

closed 2019-04-22 by nemesisdesign