openwisp/openwisp-controller #34: [x509] Auto-generated certificate object does not have organization assigned

nemesisdesign, 2018-07-20:
How to replicate:

- create a vpn-client template with `auto_cert`
- create a new device which uses the vpn template
- check the organization of the auto-generated certificate: it should be the same of the device but it's `None`

shashwat1002, 2018-11-11:
Could you please elaborate on the second point? 

> create a new device which uses the vpn template

and just to be clear

> making a vpn-client template with auto_cert
means creating vpn-client instance with `auto_cert=True` right?

nemesisdesign, 2018-11-12:
> Could you please elaborate on the second point?
> 
> > create a new device which uses the vpn template

After having created the VPN-client template, create a device, add a configuration, select the template just created and save.

> and just to be clear
> 
> > making a vpn-client template with auto_cert
> 
> means creating vpn-client instance with `auto_cert=True` right?

Create a template, select `vpn-client` as its type and ensure `auto_cert` is enabled, try and let me know.

commit 9ac2ea23 (2018-11-19): [ vpn-client] Fixed auto-certificates without organization #34
This bug happened only with shared VPNs and shared VPN-client templates.
Fixes #34
closed 2018-11-19 by nemesisdesign