openwisp/openwisp-controller #35: [geo] JS logic crashes if geometry of a location is None

nemesisdesign, 2018-09-04:
How to replicate: create a Location with no geometry set (don't place a point on the map). 
Then create a device and select the location just created and check the javascript console.

This issue is probably caused by: https://github.com/djangonauts/django-rest-framework-gis/issues/168

oltarzewskik, 2018-12-27:
Start working on it

oltarzewskik, 2018-12-29:
Does that error occurred when `openwisp-controller` was using oldest version of `djangorestframework-gis`?

nemesisdesign, 2019-01-18:
I'm not sure, have you found any interesting info about this problem?

NoumbissiValere, 2019-03-20:
Please, i wish to know if it is normal to create a location without a geometry. I thought every location should have a geometry. because it is the geometry which makes the location an actual location :thinking:

nemesisdesign, 2019-03-24:
@NoumbissiValere OpenWISP gives the possibility of flagging a location as "mobile", which means an object moves, think about a car, a bus, a train, a boat, a plane.

In this case, the location can be created with an empty geometry.

Alternatively, what we could do, which may simplify things (I only thought about this now) is to change the API which receives location data from devices to automatically create the location when the data is received, not sure if we already do that.. I'm checking, maybe I will send a PR for this myself

nemesisdesign, 2019-03-24:
I checked. Changing the model would require to change django-loci too, which is a bit tricky right now because django-loci has been migrated to channels 2, while openwisp-controller has not.

Better continue with the initial idea, I think it's safer to assume a location can be created without a geometry set only when flagged as `is_mobile == True`, because we assume we don't know the location yet

NoumbissiValere, 2020-03-05:
@nemesisdesign i found that this problem is caused by this function in `django_loci/base/admin.py/AbstractLocationAdmin`

```def json_view(self, request, pk):
        instance = get_object_or_404(self.model, pk=pk)
        return JsonResponse({
            'name': instance.name,
            'type': instance.type,
            'is_mobile': instance.is_mobile,
            'address': instance.address,
            'geometry': json.loads(instance.geometry.json)
        })```
When location has `is_mobile=True` the geometry is None. this means `instance.geometry` will return a None and `None.json` is an error. Hence the `500 error` in the javascript console 

right now am thinking two possibilities:
1) updating the above function in `django_loci` such that when `geometry is None`, it is not included in the `JsonResponse` since it is not useful (but am wondering if this will still be a problem because `django_loci` has been moved to channel 2)
2)setting a default geometry when `location is_mobile=True`. (but this doesn't make sense in real life)

Please which of the above solutions is best to implement so that i can implement it and send a PR? or should i look for another better solution?

Thanks

nemesisdesign, 2020-03-05:
@NoumbissiValere I'd fix django-loci.

NoumbissiValere, 2020-03-05:
Ok @nemesisdesign let me move to another issue then :smile:

nemesisdesign, 2020-03-05:
@NoumbissiValere what about fixing it in django-loci?

NoumbissiValere, 2020-03-05:
Yes @nemesisdesign . That's the best option i see. I wanted to fix it, but you said, you will fix. 

> @NoumbissiValere I'd fix django-loci.

Yes @nemesisdesign that's the best option i see and wanted to fix it but you said you will fix na.

nemesisdesign, 2020-03-05:
I meant that I suggest to fix it in django-loci. I'm busy on other fronts right now.

closed 2020-03-22 by nemesisdesign