openwisp/openwisp-controller #47: error 500, when clicking on device or adding new device

matar3s3, 2018-10-20:
Get Server error 500 via webinterface when clicking on name of a device or click add device under
"/admin/config/device/"

openwisp was installed via the documentation with vagrant on virtualbox 
https://github.com/openwisp/vagrant-openwisp2

thanks Stefan

error.log

 File "/opt/openwisp2/env/lib/python3.5/site-packages/django/template/base.py", line 937, in render
    bit = node.render_annotated(context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/template/base.py", line 904, in render_annotated
    return self.render(context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/template/defaulttags.py", line 209, in render
    nodelist.append(node.render_annotated(context))
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/template/base.py", line 904, in render_annotated
    return self.render(context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/template/defaulttags.py", line 209, in render
    nodelist.append(node.render_annotated(context))
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/template/base.py", line 904, in render_annotated
    return self.render(context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/template/defaulttags.py", line 309, in render
    return nodelist.render(context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/template/base.py", line 937, in render
    bit = node.render_annotated(context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/template/base.py", line 904, in render_annotated
    return self.render(context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/template/defaulttags.py", line 309, in render
    return nodelist.render(context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/template/base.py", line 937, in render
    bit = node.render_annotated(context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/template/base.py", line 904, in render_annotated
    return self.render(context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/template/base.py", line 993, in render
    return render_value_in_context(output, context)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/template/base.py", line 972, in render_value_in_context
    value = str(value)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/utils/html.py", line 397, in <lambda>
    klass.__str__ = lambda self: mark_safe(klass_str(self))
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/forms/boundfield.py", line 32, in __str__
    return self.as_widget() + self.as_hidden(only_initial=True)
  File "/opt/openwisp2/env/lib/python3.5/site-packages/django/forms/boundfield.py", line 93, in as_widget
    renderer=self.form.renderer,
TypeError: render() got an unexpected keyword argument 'renderer'

nemesisdesign, 2018-10-21:
I found out a possible reason why this happens.

The current requirements.txt (which are read by `setup.py`) doesn't specify any django vesion. The only dependency that seems to specify the django version is django-model-utils which specifies `django>=1.81` which in turn is causing the latest version of django available to be installed.

nemesisdesign, 2018-10-21:
I correct myself, It's actually django-loci: https://github.com/openwisp/django-loci/blob/master/requirements.txt#L1

nemesisdesign, 2018-10-21:
I correct myself again, it's caused by a line in ansible-openwisp2 that installs django-redis.

The version of django should be set by django-netjsonconfig (which is the base module on which openwisp-controller is based) but it's being overridden by this line in ansible-openwisp2: https://github.com/openwisp/ansible-openwisp2/blob/ef9ee14ef762b4b98dc0f84fe9ee19ccb8197823/tasks/pip.yml#L50
Which basically calls `pip install --upgrade django-redis` which in turns upgrades to the last django version available and overrides the django version defined in the dependencies of openwisp-controller.

nemesisdesign, 2018-10-22:
@matar3s3 we released a fix of the ansible-openwisp2 on ansible-galaxy, you may want to upgrade as described here: https://github.com/openwisp/ansible-openwisp2#upgrading-openwisp2