openwisp/openwisp-controller #50: [bug] 500 error when adding a new device

dwang, 2018-10-25:
While adding a new device to the web UI, I experienced the same issue in #24 which was because of not selecting an organization and saving the config.

I did some more testing and found that you also get a 500 error if you select `New` for `Location selection`, but do not select the `Type`. Example:

[image]

Here are the errors from `/opt/openwisp2/log/error.log`:

```
[ERROR 2018-10-25 22:06:15,594] module: exception, process: 28929, thread: 140151813760768
Internal Server Error: /admin/geo/location/floorplans/json/
Traceback (most recent call last):
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/db/models/fields/__init__.py", line 2363, in to_python
    return uuid.UUID(value)
  File "/usr/lib/python3.6/uuid.py", line 140, in __init__
    raise ValueError('badly formed hexadecimal UUID string')
ValueError: badly formed hexadecimal UUID string

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/core/handlers/exception.py", line 35, in inner
    response = get_response(request)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/core/handlers/base.py", line 128, in _get_response
    response = self.process_exception_by_middleware(e, request)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/core/handlers/base.py", line 126, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/utils/decorators.py", line 142, in _wrapped_view
    response = view_func(request, *args, **kwargs)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/views/decorators/cache.py", line 44, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/contrib/admin/sites.py", line 223, in inner
    return view(request, *args, **kwargs)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django_loci/base/admin.py", line 74, in json_view
    instance = get_object_or_404(self.model, pk=pk)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/shortcuts.py", line 87, in get_object_or_404
    return queryset.get(*args, **kwargs)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/db/models/query.py", line 397, in get
    num = len(clone)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/db/models/query.py", line 254, in __len__
    self._fetch_all()
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/db/models/query.py", line 1182, in _fetch_all
    self._result_cache = list(self._iterable_class(self))
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/db/models/sql/compiler.py", line 1055, in execute_sql
    sql, params = self.as_sql()
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/db/models/sql/compiler.py", line 461, in as_sql
    where, w_params = self.compile(self.where) if self.where is not None else ("", [])
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/db/models/sql/compiler.py", line 393, in compile
    sql, params = node.as_sql(self, self.connection)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/db/models/sql/where.py", line 80, in as_sql
    sql, params = compiler.compile(child)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/db/models/sql/compiler.py", line 393, in compile
    sql, params = node.as_sql(self, self.connection)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/db/models/lookups.py", line 161, in as_sql
    rhs_sql, rhs_params = self.process_rhs(compiler, connection)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/db/models/lookups.py", line 260, in process_rhs
    return super().process_rhs(compiler, connection)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/db/models/lookups.py", line 93, in process_rhs
    return self.get_db_prep_lookup(value, connection)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/db/models/lookups.py", line 187, in get_db_prep_lookup
    [get_db_prep_value(value, connection, prepared=True)]
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/db/models/fields/__init__.py", line 2354, in get_db_prep_value
    value = self.to_python(value)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/db/models/fields/__init__.py", line 2368, in to_python
    params={'value': value},
django.core.exceptions.ValidationError: ["'floorplans' is not a valid UUID."]

[ERROR 2018-10-25 22:14:11,047] module: exception, process: 28929, thread: 140152300586176
[... 83 lines cut ...]
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/utils/decorators.py", line 58, in bound_func
    return func.__get__(self, type(self))(*args2, **kwargs2)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/contrib/admin/options.py", line 1451, in changeform_view
    return self._changeform_view(request, object_id, form_url, extra_context)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/contrib/admin/options.py", line 1490, in _changeform_view
    if all_valid(formsets) and form_validated:
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/forms/formsets.py", line 442, in all_valid
    if not formset.is_valid():
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/forms/formsets.py", line 301, in is_valid
    self.errors
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/forms/formsets.py", line 281, in errors
    self.full_clean()
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/forms/formsets.py", line 329, in full_clean
    form_errors = form.errors
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/forms/forms.py", line 174, in errors
    self.full_clean()
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/forms/forms.py", line 376, in full_clean
    self._clean_fields()
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django/forms/forms.py", line 397, in _clean_fields
    value = getattr(self, 'clean_%s' % name)()
  File "/opt/openwisp2/env/lib/python3.6/site-packages/django_loci/base/admin.py", line 187, in clean_floorplan
    if self.cleaned_data['type'] != 'indoor' or self.cleaned_data['floorplan_selection'] == 'new':
KeyError: 'type'
```
If needed, I can look into both cases further and possibly create a fix.

dwang, 2018-10-26:
I have no preference but I guess I can start with the unselected organization first because the issue was created earlier.

Thanks!

dwang, 2018-10-29:
After looking into this more, the unselected organization 500 error does not look easy to reproduce locally. I only got the error to trigger once. On the other hand, I can reproduce it easily on a server created with `ansible-openwisp2`.

I'll try again tomorrow.

nemesisdesign, 2018-11-02:
Thx for the update @dwang

JayjeetAtGithub, 2019-01-13:
Is this issue open ?

NoumbissiValere, 2019-02-01:
I think the Location error can be solved by making sure that when ever a device has a location instance, this instance should have a **type** value before saving it else raise a ValidationError. 
I will have a look at this

NoumbissiValere, 2020-03-05:
@nemesisdesign this issue was caused by an unhandled exception from `django_loci` and you addressed it when you merged [this pull request](https://github.com/openwisp/django-loci/pull/45). 
So i think the issue should be closed

nemesisdesign, 2020-03-05:
I tested this again and confirm those issues are fixed, thanks @NoumbissiValere for the ping.

closed 2020-03-05 by nemesisdesign