openwisp/openwisp-controller #56: AttributeError: module 'django_netjsonconfig.settings' has no attribute 'BACKEND_DEVICE_LIST'

sah-anshu, 2018-11-04:
Hi,
I am unable to start docker. please help.

  File "/usr/local/lib/python3.6/site-packages/django/utils/autoreload.py", line 248, in raise_last_exception
    raise _exception[1]
  File "/usr/local/lib/python3.6/site-packages/django/core/management/__init__.py", line 327, in execute
    autoreload.check_errors(django.setup)()
  File "/usr/local/lib/python3.6/site-packages/django/utils/autoreload.py", line 225, in wrapper
    fn(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/django/__init__.py", line 24, in setup
    apps.populate(settings.INSTALLED_APPS)
  File "/usr/local/lib/python3.6/site-packages/django/apps/registry.py", line 120, in populate
    app_config.ready()
  File "/usr/local/lib/python3.6/site-packages/django/contrib/admin/apps.py", line 23, in ready
    self.module.autodiscover()
  File "/usr/local/lib/python3.6/site-packages/django/contrib/admin/__init__.py", line 26, in autodiscover
    autodiscover_modules('admin', register_to=site)
  File "/usr/local/lib/python3.6/site-packages/django/utils/module_loading.py", line 47, in autodiscover_modules
    import_module('%s.%s' % (app_config.name, module_to_search))
  File "/usr/local/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/local/lib/python3.6/site-packages/openwisp_controller/config/admin.py", line 35, in <module>
    class DeviceAdmin(MultitenantAdminMixin, AbstractDeviceAdmin):
  File "/usr/local/lib/python3.6/site-packages/openwisp_controller/config/admin.py", line 41, in DeviceAdmin
    if django_netjsonconfig_settings.BACKEND_DEVICE_LIST:
AttributeError: module 'django_netjsonconfig.settings' has no attribute 'BACKEND_DEVICE_LIST'

okraits, 2018-11-05:
Which version of openwisp-controller are you using? 

`BACKEND_DEVICE_LIST` currently requires the latest development version of openwisp-controller because that feature is not released yet.

sah-anshu, 2018-11-06:
I have now used old version & its working :)
Thanks

closed 2018-11-06 by sah-anshu