openwisp/openwisp-controller #58: How to add ALLOWED_HOSTS to settings.py in Docker

sah-anshu, 2018-11-06:
Hi,

I tried to implement openwisp-controller in Docker and when i am running the application, It is giving  me error. 

System check identified no issues (0 silenced).
November 05, 2018 - 00:47:12
Django version 2.0.9, using settings 'settings'
Starting Channels development server at http://0.0.0.0:8000/
Channel layer default (asgiref.inmemory.ChannelLayer)
Quit the server with CONTROL-C.
2018-11-05 00:47:12,836 - INFO - worker - Listening on channels http.request, websocket.connect, websocket.disconnect, websocket.receive
2018-11-05 00:47:12,838 - INFO - worker - Listening on channels http.request, websocket.connect, websocket.disconnect, websocket.receive
2018-11-05 00:47:12,839 - INFO - worker - Listening on channels http.request, websocket.connect, websocket.disconnect, websocket.receive
2018-11-05 00:47:12,839 - INFO - worker - Listening on channels http.request, websocket.connect, websocket.disconnect, websocket.receive
2018-11-05 00:47:12,840 - INFO - server - HTTP/2 support not enabled (install the http2 and tls Twisted extras)
2018-11-05 00:47:12,841 - INFO - server - Using busy-loop synchronous mode on channel layer
2018-11-05 00:47:12,841 - INFO - server - Listening on endpoint tcp:port=8000:interface=0.0.0.0
Invalid HTTP_HOST header: 'auth.infi.zone:8000'. You may need to add 'auth.infi.zone' to ALLOWED_HOSTS.
[2018/11/05 00:47:17] HTTP GET / 400 [0.08, 58.84.14.10:26959]

[image]

Please advice. 

Regards
Anshu

nemesisdesign, 2018-11-06:
Try adding `local_settings.py` in `tests/`, should work. Keep in mind this is intended only for development.

sah-anshu, 2018-11-06:
Yes , for development it is OK, but how can i implement .env files in Settings.py while using Docker.  I suppose we need to modify settings.py accordingly. But i can't locate it

nemesisdesign, 2018-11-06:
it's in `test/settings.py` but again, that `Dockerfile` we have here is not intended for production usage. For production we have an ansible role at https://github.com/openwisp/ansible-openwisp2.
Dockerizing openwisp2 is not as trivial as this `Dockerfile` we have here, especially with the new version which should be released in a couple of months.

closed 2018-11-06 by nemesisdesign