openwisp/openwisp-controller #63: [enhancement] devices installed on floor plan

serra82, 2018-11-15:
Is possible to see all devices installed on a selected floor plan?
Actually is possible to see only the position of the selected devices, but could be very usefuel see all devices installed on a floor plan to plan upgrade and covered wifi areas.
thank you

atb00ker, 2019-11-19:
> Is possible to see all devices installed on a selected floor plan?
> Actually is possible to see only the position of the selected devices, but could be very usefuel see all devices installed on a floor plan to plan upgrade and covered wifi areas.

Unfortunately, no way right now!

(Closing this for now, please feel free to reanimate if required.)

Bixilon, 2021-06-18:
Damn I need this

atb00ker, 2021-06-18:
Reopening as feature request instead of a question. 😄

nemesifier, 2023-10-25:
We did some work in the frontend library to implement the floorplan map: 
https://openwisp.github.io/netjsongraph.js/examples/netjsonmap-indoormap.html

We need to implement the backend part and integrate.