openwisp/openwisp-controller #65: [vpn-server] Having 2 VPNs that share certificate and CA triggers bug in device configuration

nemesisdesign, 2018-11-23:
Reusing the same CA and VPN certificate in 2 VPNs and activating both on a device triggers a validation error in [netjsonconfig](https://github.com/openwisp/netjsonconfig). It's most likely a bug that needs to be fixed first in netjsonconfig.

atb00ker, 2020-03-06:
@nemesisdesign I just created a device with 2 vpn-clients from different VPN servers which are sharing CA and cert and it worked.
However, I am not aware in which commit it was fixed!

nemesisdesign, 2020-03-06:
@atb00ker in this one on netjsonconfig: https://github.com/openwisp/netjsonconfig/commit/d451e6224c38425a8dd10b68c31423bdbac829df

atb00ker, 2020-03-07:
@nemesisdesign but it was never merged, notice it's still `True` in the master : https://github.com/openwisp/netjsonconfig/blob/master/netjsonconfig/schema.py#L184 !

nemesisdesign, 2020-03-07:
Oh you're right. I thought it was merged. Why is it so, can you send it?

atb00ker, 2020-03-09:
Sure, on it! :smile:

atb00ker, 2020-03-09:
This should solve it : https://github.com/openwisp/django-netjsonconfig/pull/160

closed 2020-03-13 by nemesisdesign