openwisp/openwisp-controller #67: Network and VLAN objects?

epoch1970, 2018-11-28:
New user here. I went through a painful installation (apt-get, pip, pipenv yadda yadda yadda) to see the controller GUI in action. It werks! (v0.3.2)

I find the idea of NetJSON compelling but I am puzzled not to have found a "network" object nor the notion of VLAN ID for devices. I expected to find them in the GUI. So I went and checked the spec on netjson.org, but found nothing there either.

Am I missing something?

Thanks in advance.

nemesisdesign, 2018-11-28:
What do you mean for "network" object?

The NetJSON spec is still an ongoing effort, it only covers the basics and there are still things which are missing.

For VLAN ID, do you mean the programmable switch feature of OpenWRT?

Or do you mean defining interface VLANs? This can be done by defining an interface named something like `<name>.<vlan-id>`.

I also suggest looking for "vlan" in this page: http://netjsonconfig.openwisp.org/en/latest/backends/openwrt.html

epoch1970, 2018-11-29:
Thank you for your answer, Federico. Openwrt/LEDE is not part of my background nor my target application, your pointer to the 'switch' OpenWRT object is useful.

As for the 'network object’: One of the goals of NetJSON's 'network collection' is to "list devices of a network”. 
I think I would like to “list networks of a device”: my use case is a linux “edge device”, running containers, capable of working with peers (local vlans/wifi networks). At network config time it needs to move around local network addresses to avoid clashing with the site's network (eg DHCP uplink); A list of the live/expected networks would be useful to me, I think.

Does this make sense? I defer to your opinion/advice. Thanks again.