openwisp/openwisp-controller #76: [models] Device per organization unique name & mac_address

nemesisdesign, 2018-12-14:
Right now a device must have a unique name and a unique mac address across all organization but this is not right.

Different organizations may use the same name for a device and they should not know that another organization is using the same name.

Also, a device may be moved from one organization to another for testing reasons and I don't see a good reason to prevent that. 

So I think we should remove the unique constraint from:

- `name`
- `mac_address`

And add a `unique_together` constraint for:

- `name` and `organization`
- `mac_address` and `organization`

okraits, 2018-12-14:
@nemesisdesign I think the same applies then to the hardware id/serial number, right?

nemesisdesign, 2018-12-15:
@okraits yes you're right, I forgot about that, thanks for reminding me, I'll add it to the issue description

PR #77 (2019-01-17): [models] Device per organization unique name & mac_address #76
@nemesisdesign , I didn't overshadow `hardware_id` because it designed such that user decides what the `unique` constraints are, but I put the `unique_together` for it anyway so that it'll work whether the `unique` constraint is `True` or not. 
This code also adds respective tests.
Fixes #76
commit 0f38f9bc (2019-01-17): [models] Device per organization unique name & mac_address #76

Closes #76
closed 2019-01-17 by nemesisdesign