openwisp/openwisp-controller #82: [docker] Missing requirements.txt file crashing build

prusayn, 2019-01-23:
Because of specification of base image python:3-onbuild it need requirements.txt. Oherwise it cause error and don't allow to build image.

Easiest fix is to just create empty requirements.txt file in project root dir. It would be nice to move dependencies in Dockerfile:
```
pip3 install -U pip setuptools wheel
````
to requirements file.

acyuta108, 2019-04-04:
@prusayn why don't you create a pull request?

atb00ker, 2019-11-19:
It's outdated now because we have moved to `pipenv` in this repository now.
I am afraid that the docker image needs to be made from stratch now. 
Some work is done here: https://github.com/openwisp/openwisp-controller/pull/111

(Closing this for now, please feel free to reanimate if required.)

closed 2019-11-19 by atb00ker