openwisp/openwisp-controller #83: [enhancement]MAC address input field

vedangj044, 2019-01-26:
We are using a plain text field for entering the "mac address" text field in the "device/add' section. I think we can override the default template by dividing the the text field in six different text fields, which can make entering the mac address easier. .It requires a very basic script. Here, is the code.

```
<form method="post">
    <input id='1' style="width:20px" type="text" maxlength="2" onkeyup="move(this, '2')">
    <input id='2' style="width:20px" type="text" maxlength="2" onkeyup="move(this, '3')">
    <input id='3' style="width:20px" type="text" maxlength="2" onkeyup="move(this, '4')">
    <input id='4' style="width:20px" type="text" maxlength="2" onkeyup="move(this, '5')">
    <input id='5' style="width:20px" type="text" maxlength="2" onkeyup="move(this, '6')">
    <input id='6' style="width:20px" type="text" maxlength="2">

    <script type="text/javascript">
      function move(from, to){
        var length = from.value.length;
        var max = from.getAttribute('maxlength');
        if(length == max){
          var next = document.getElementById(to);
          next.focus();
        }
      }
    </script>
</form>

atb00ker, 2019-01-29:
Hello @vedangj044 ,

Can you please point out benefits of the change or problem faced by you while entering mac address in the corrent system? :smile: 

Thanks!

vedangj044, 2019-01-29:
Hello @atb00ker,

When we need to enter the Mac address we need to use ' : ' (colon), rather if we split the input into separate input fields it would be easier.

NoumbissiValere, 2019-01-29:
meaning we should have 6 inputs instead of one?

vedangj044, 2019-01-29:
Yes, @NoumbissiValere i guess i can explain my idea more easily with this 
http://vedangj.tech/work.html

nemesisdesign, 2019-01-30:
In most cases the mac address is not entered manually but it is automatically sent during auto-registration, which is the recommended way to add devices into the system, therefore the complexity needed to implement this change won't bring any benefit to our users, so in order to keep the complexity at a sane level we should not move on with this.

Thanks for proposing this idea.

closed 2019-01-30 by nemesisdesign