openwisp/openwisp-controller #89: Use x-forwarded-for field for last IP

rudihorn, 2019-03-31:
Currently OpenWISP just uses the last IP address that established communication with the device. I have a setup which has OpenWISP is set up on a server, which has it's connections forwarded by an HTTP proxy. The HTTP proxy sets the X-Forwarded-For header which should contain the actual IP address, meaning that the device Last IP is set to the IP of the HTTP proxy rather than the device.

The correct behaviour would be to use the X-Forwarded-For header if it is set.

nemesisdesign, 2019-03-31:
Hi @rudihorn, I think you have to improve the configuration of your frontend HTTP server, see the nginx configuration of the official installer to see how it's done:
https://github.com/openwisp/ansible-openwisp2/blob/master/templates/nginx.j2

So I'm closing this issue because it's not an issue with this module but rather with how the deployment is done.

closed 2019-03-31 by nemesisdesign

rudihorn, 2019-04-01:
I have now ended up reconfiguring it so I use the uwsgi proxy rather than an HTTP proxy. Looking at the code this would be a Django bug in any case though, so it is fine to close this.

nemesisdesign, 2019-04-01:
I'm glad that you solved it :+1: