openwisp/openwisp-controller #90: [enhancement] Ability to set device passwords

rudihorn, 2019-03-31:
Currently the network device templates don't have an option to set the root password. I think this would be something helpful to have for centralised device management.

nemesisdesign, 2019-03-31:
That would mean having to regenerate the `/etc/shadow` file, the system would need to know its contents before editing it.

It can be done, but I'm wondering why spend time to add complexity when you can set the password during compilation time (ensuring also that no unauthorized person can put the device in failsafe mode and enter via SSH without password) and prepare a template with SSH keys so a specific group of authorized users can log in without password.

rudihorn, 2019-04-01:
This means that it is necessary to configure a custom OpenWRT image just for the purpose of managing the password, and if one has a fleet of devices it is either necessary to change the password on each individual device or to reflash all of the devices. OpenWISP essentially has full control over the device in any case, so it may as well have the power to configure the password. I think it would also be nice if it was possible to use out of the box precompiled OpenWRT images, and to then be able to commission devices with as little effort as possible (install package, enter url). Every small other step that the user may have to configure manually for each device on initial setup (which might already be weird network issues) just makes it more tedious.

nemesisdesign, 2019-04-01:
It's not that it is necessary to compile a custom OpenWRT image just to set the password. You have to do that for different good reasons and one is security, if you don't set a default password at firmware level, your network is insecure because anyone can unplug a device, put it in failsafe mode, SSH into it, set the password, reboot it and then log in via SSH and analyze the internal configuration of your network.

You can set the password with the current version of OpenWISP by preparing an `/etc/shadow` file, but I really do not recommend doing it.

Password should be set at firmware level at compile time with a very strong password, login should be performed with SSH keys and not with password.

Regarding pre-compiled firmware images: that can be done with the OpenWRT image-generator.

rudihorn, 2019-04-03:
I have for the moment opted for the solution you have described. I still feel like it is worth still having the setting to set passwords, as otherwise it would still be necessary to reflash all devices when one does want to do so. Depending on the device they aren't exactly very secure without encryption anyways, as most OpenWRT routers can be opened and it is possible to boot a custom initramfs using a serial connection. I agree that if one does have the option to set the password a warning would be in order.

nemesisdesign, 2020-08-31:
This feature will be implemented in https://github.com/openwisp/openwisp-controller/issues/253.

pandafy, 2021-04-06:
This has been implemented in #421. I encourage stakeholders to help us test it.

closed 2021-04-06 by pandafy

nemesisdesign, 2021-06-02:
https://github.com/openwisp/openwisp-controller/pull/421 is merged!