openwisp/openwisp-controller #91: [enhancement] device version

rudihorn, 2019-03-31:
Currently the devices don't report their version information. This is helpful to know which devices to upgrade etc. There is a field for `operating system` but it is unclear if this should be the version / build as well and I would probably suggest adding another field for the revision ID.

It should be easy to get this information on OpenWRT. There is a file `/etc/openwrt_release` which has contents that look like:

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r9639-57b1e951b5'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r9639-57b1e951b5'
DISTRIB_TAINTS='no-all'
```

To retrieve the Operating System field it would be possible to just execute:

`. /etc/openwrt_release; echo "$DISTRIB_ID"`

The version / revision can be retrieved with:

`./etc/openwrt_release; echo "$DISTRIB_RELEASE $DISTRIB_REVISION"`

Or possibly the operating system could be the ID + RELEASE and the version could just be the revision?

nemesisdesign, 2019-03-31:
What version (commit hash) are you using?
This is already implemented, see the following screenshot.
[image]

closed 2019-03-31 by nemesisdesign

rudihorn, 2019-04-01:
Hmm odd, so my version has these fields but they aren't filled out and it wasn't obvious to me that Operating System should also include the version number. If this is implemented then that is great, but this probably means my setup is broken somewhere. 

My OpenWISP controller should be the latest version of PIP (0.3.2) and has the fields, but they are just empty. My OpenWRT was recently (14 days or so) compiled using the master branch of OpenWRT with openwisp fetched from the feeds.

nemesisdesign, 2019-04-01:
I suggest you to follow the instructions in our documentation, the package installed from the official feed is not up to date but there's no point in our docs in which we suggest to install from there.

See here:

- http://openwisp.io/docs/user/configure-device.html#install-openwisp-config-on-your-openwrt-instance
- http://openwisp.io/docs/user/configure-device.html#compile-your-own-openwrt-image

rudihorn, 2019-04-03:
Hmm so I have made sure to get the latest version from the official feed, and this seems to work for newly provisioned devices, but previously provisioned devices are not updating their version numbers despite having the latest version installed. I have also tried factory resetting them so they connect to the controller with the shared key, but seemingly this doesn't make them update the fields.

nemesisdesign, 2019-04-03:
you have to make the device register again, remove the uuid and key from the config file, add the shared secret and restart the deamon.

We should have this information updated over time though, so there's an issue in openwisp-config to track this: https://github.com/openwisp/openwisp-config/issues/58
But it requires changing the controller as well.

rudihorn, 2019-04-04:
Yes I was wondering if openwisp/openwisp-config#58 was an issue as well, but if that is already reported then that is fine.