openwisp/openwisp-controller #95: [model] modify the template model.

NoumbissiValere, 2019-05-15:
This issue is related to #112 .
Modify the template model to enable users specify the URL from which they will import their templates from.

nemesisdesign, 2019-05-16:
You need to make a list of all the modifications that are planned here. Make the issues specific, generic information is not usable. Issues need to be actionable, so we can clearly understand if they were completed successfully or not.

NoumbissiValere, 2019-05-17:
I think an `update` flag should be added to the template model. this boolean will be used as follows:
When an update is available for an external template, this is set to `True`  and `False` otherwise. this value will be checked each time the external template `change_page`  is visited so as to notify the user if an update is available or not.

closed 2019-08-17 by NoumbissiValere