openwisp/openwisp-controller #97: [model] Add Notification model

NoumbissiValere, 2019-05-15:
Add a base notification model which will have fields;
1)a template field which will be a foreign key from the Template Model
2) Subscriber domain name which will store the domain name of the admin who subscribed to a template in the `Subscribe` model. This will equally store the domain name of the person who unsubscribed from a template in the `Unsubscribe`  model. 
3) Date field which will store the date when the admin subscribed to a particular template
4) Modify field which will store the last date when the subscriber synchronized its template to the latest template version.

Create a `Subscribe` and `Unsubscribe` models which will inherit from the base `Notification` model and do necessary modifications

nemesisdesign, 2019-05-16:
`Notification` is not a good name, `TemplateSubscription` is the first good alternative, I don't think we need two models to do this.

NoumbissiValere, 2019-05-16:
I wanted to create separate subscribe users from unsubscribed users.
But it's ok, i will use one table and add a Boolean field which will be `True` if a user subscribed and `False` otherwise.

closed 2019-08-17 by NoumbissiValere