openwisp/openwisp-controller #103: Docker build error

vikrant9305, 2019-05-28:
When building openwisp controller (cloned current working version), the below error is thrown after Step 1: FROM python:3-onbuild

COPY failed: stat /var/lib/docker/tmp/docker-builder712687534/requirements.txt: no such file or directory

*something like problem of building it with python3-onbuild.
[image]

atb00ker, 2019-05-28:
Ah, `requirements.txt` doesn't exist. We have moved to pipenv.
This needs to be updated as well! 
Thanks for opening an issue for it. :+1:

vikrant9305, 2019-05-31:
Your comment in #108 
"_Just for my information: cd openwisp-controller/ && pip3 install -U works right?_"

No. To do "cd openwisp-controller/" we need to put the source in docker. That's why I have done "git clone" changes in dockerfile in my repository.

Let me know once #107 resolved so i can merge my Dockerfile changes to original repo by initiating new pull request.

frextrite, 2019-11-07:
For anyone stumbling on this, this is being worked on in #111

atb00ker, 2019-11-08:
The dockerfile in the mentioned PR #111 is working right now to the best of my knowledge, if it doesn't work please feel free to leave a comment! :smile:

nemesisdesign, 2019-11-29:
Closed by #111. Thanks @atb00ker :+1:

closed 2019-11-29 by nemesisdesign