openwisp/openwisp-controller #104: [ux] errors made while adding a new device get hidden

atb00ker, 2019-05-30:
Description:

When adding a new device if someone makes an error of not entering `Geometry` for a `Outdoor environment`, the user will `Please correct the errors below.` at the top of the page but the actual field to make the correction is not shown.
The user needs to change the `Type` field then select `Outdoor environment` again to see the errored field.
This may cause confusion to users.

oltarzewskik, 2019-08-21:
I don't know `jquery`, but I found that you have do something similar for `outdoor` like it was done here:
https://github.com/openwisp/django-loci/blob/master/django_loci/static/django-loci/js/loci.js#L328-L332

muarachmann, 2019-11-11:
@atb00ker can I take on this, from what I guess i will need to include the appropriate js code in the `change_form` template and then make the section visible. Let me know if I am on track

atb00ker, 2019-11-20:
@muarachmann I can't be sure of the approach without looking at the pull request, but if you want to take it, go for it!

muarachmann, 2019-11-20:
Okay

marfgold1, 2019-12-05:
Hello, I did some research and found that [the code that @oltarzewskik said](https://github.com/openwisp/django-loci/blob/master/django_loci/static/django-loci/js/loci.js#L328-L332) isn't actually the best answer, since the required field isn't just .field-indoor (User can actually set the indoor coordinates and floor, but leaving the geographic information empty. When user save, the user still will get error `Please correct the errors below.` at the top of the page but the actual field to make the correction is not shown, and somehow the Floorplan selection is showed, where it shouldn't be appeared).

atb00ker, 2019-12-06:
@marfgold1 I can't be 100% sure of the solution without investigating the bug but what solution do you think would work?

marfgold1, 2019-12-06:
@atb00ker I just found the perfect solution! We just have to `typeChange` to `$type.val()` when the document is ready! This also prevent another UX bug (the one that I mentioned before). I will create a pull request for this.

marfgold1, 2019-12-06:
@atb00ker also I've found the bug when `type` is selected and `isMobile` is toggled on. Should I create an issue for this, or create another pull request, or just add another commit in the same pull request?