openwisp/openwisp-controller #107: "The Travis CI build failed" #106

vikrant9305, 2019-05-30:
Tried to make [a pull request](https://github.com/openwisp/openwisp-controller/pull/106) but got [this error](https://travis-ci.org/openwisp/openwisp-controller/builds/539266129/config).

atb00ker, 2019-05-30:
Hi,

I just re-ran travis on master, it worked. So it doesn't seem like an issue due a bug rather we need to checkout the pull request itself.
We can discuss the errors faced while making a PR in the pull request itself, so in order to avoid the confusion of a conversation being in multiple places, I am closing this issue. :smile:

atb00ker, 2019-05-30:
Tests failed now on master due to the same reason, It seems like something needs to be updated in the travis-ci. Reanimating this issue with a new header.

atb00ker, 2019-11-20:
Fixed: https://github.com/openwisp/openwisp-controller/pull/109

closed 2019-11-20 by atb00ker