openwisp/openwisp-controller #108: Very old Dockerfile

vikrant9305, 2019-05-30:
openwwisp-controller installation in docker failed on step "pip3 install -U ." as there is no setup.py file at location "." in docker.

modification for the same has been done at [my repository](https://github.com/vikrant9305/openwisp-controller/blob/master/Dockerfile).
[image]

atb00ker, 2019-05-30:
Just for my information: `cd openwisp-controller/ && pip3 install -U` works right?

atb00ker, 2019-05-30:
Same here. This is a duplicate issue because both require the dockerfile to be fixed because it's outdated because we moved to `Pipenv` for this repository. So in order to avoid the conversation being at multiple places, I am closing this one. Let's discuss in the comments of the original issue: https://github.com/openwisp/openwisp-controller/issues/103

closed 2019-05-30 by atb00ker