openwisp/openwisp-controller #121: Configuration of Wireless

toghraee, 2019-09-04:
Hi

I'm new to OpenWISP and found it very useful. I'm building a test environment with few OpenWRT wireless Access points (BT Hub 5 Type A) and OpenWISP2 

I have successfully deployed the OWISP2 on a Raspberry Pi using the Ansible module and have the OWISP2 up and working.
I managed to installed the openwisp-config-openssl_0.4.6a-1_all.ipk on the OpenWRT device , did the configuration on the ini file to point to the OWISP2 server with correct Shared Key, SSL verify set to NO .
The AP successfully registered to OWISP and I was able to create some basic templates for assigning the DNS.

My aim is to create a template to manage the WiFi SSID , password. 

my OpenWRT devices already have (existing) : 

Radios:
 radio0 (5ghz) 
 radio1 (2ghz) 

Interfaces : 
 wlan0 : Default SSID for 5ghz radio0
 wlan1 : Default SSID for 2ghz radio0
 br-lan (LAN)  : which is a bridge for all RJ45 ports and the wlan0 and wlan1 interfaces and has a static IP address for management
 WAN : Not Used
  ADSL: Not used

I tried to create a Wireless Interface in OWISP2 template,  tried calling it "wlan5" and  assigned it to the network "LAN(br-lan)"  but after the openwrt downloads and applies the configuration it loses its IP address and I think it somehow destrys the existing br-lan. 
in this situation, I can only revert to factory setting by pressing the reset button on the device.

may I please seek help to understand what is the best way to create and manage SSIDs (wireless interfaces) with OWISP and such existing configuration on OpenWRT?

Thanks
Reza

atb00ker, 2019-11-19:
Hello, 

May I ask, do you suspect it to be a bug in openwisp-controller? 
( I haven't tried reproducing it, so just want to know if you have done any investigation. )

okraits, 2019-11-19:
Did you try to remove the wireless interface in the template? If you remove any configuration for that device on the server, then the config on the device should be reverted to the state before the config update.

nemesisdesign, 2019-11-19:
Closing as not a bug report. Please use the support channels (eg: mailing list).

closed 2019-11-19 by nemesisdesign