openwisp/openwisp-controller #122: TemplateDoesNotExist

rajatarora21, 2019-11-02:
Hey I'am new to Openwisp and tried to clone the project. Though when I run the project , it gives me an error that templates doesn't exist . I think they have been moved to some other location.
Please let me know the new location , so that I can try to fix the bug. 
[image]

IAmAMissingColon, 2019-11-02:
Hello! Can You Tell What's The Trackback In Terminal ?

IAmAMissingColon, 2019-11-02:
Can You Describe What Are You Doing ?

rajatarora21, 2019-11-02:
Hey there ! I just followed the documentation given in README for my installation, when I run the server locally using pipenv run ./manage.py runserver 0.0.0.0:8000 , the server is running good , but during the run django is not able to locate the source of the templates(all your static files). I tried to go to the location using .But the location doesn't exist .

[image]

IAmAMissingColon, 2019-11-02:
pipenv?

rajatarora21, 2019-11-02:
Yeah i am using pipenv,(as given in the documentation).Check out the README.rst

rajatarora21, 2019-11-02:
cd /home/YOURUSERNAME/.local/share/virtualenvs/openwisp-controller-Ut1Pu8_1/lib/python3.6/site-packages/django/contrib/auth/templates/django/forms/widgets/.
Can you run this on your terminal , and check if all your templates are in widget folder

IAmAMissingColon, 2019-11-02:
Ok, My Bad !

Wait!

rajatarora21, 2019-11-02:
cd /home/rajat/.local/share/virtualenvs/openwisp-controller-Ut1Pu8_1/lib/python3.6/site-packages/django/forms/templates/django/forms/widgets/
I finally found the correct directory for the templates , using locate in terminal.
I suggest to fix this bug , so that later someone doesn't face the same error.

atb00ker, 2019-11-02:
Hello, do you get this error even after running: `pipenv run install_dev` as mentioned in the README?

IAmAMissingColon, 2019-11-02:
> Hello,
> 
> Thanks for sharing this.
> Do you get this error even after running: `pipenv run install_dev` as mentioned in the README?

I'm Getting It After Running `pipenv run install_dev`

rajatarora21, 2019-11-03:
You're Welcome.
Yeah ,That's what I was checking earlier that i didn't miss any commands while cloning, But the error still popped up even after running all the commands.

oltarzewskik, 2019-11-05:
@rajatarora21 Can you give me output of this command: `lsof -i`, when `openwisp-controller` is not running.

atb00ker, 2019-11-20:
I am unable to reproduce this, it may because of incorrect setup, please follow https://github.com/openwisp/openwisp-controller#installing-for-development incase it's still not fixed, use the mailing list to share about the setup steps and system information to help us reproduce this!

closed 2019-11-20 by atb00ker