openwisp/openwisp-controller #123: Package conflict while installing using pipenv

frextrite, 2019-11-07:
```
ERROR: django-netjsonconfig 0.9a0 has requirement django-x509<0.5.0,>=0.4.0, but you'll have django-x509 0.5a0 which is incompatible.
ERROR: django-netjsonconfig 0.9a0 has requirement openwisp-utils<0.3,>=0.2, but you'll have openwisp-utils 0.3a0 which is incompatible.
ERROR: django-loci 0.2.1 has requirement openwisp-utils<0.3.0,>=0.2, but you'll have openwisp-utils 0.3a0 which is incompatible.
```

Python 3.5.9

IAmAMissingColon, 2019-11-07:
Yup, Having Same Error

atb00ker, 2019-11-08:
Two questions: 

1. Are you installing for development?

2. If django is working fine, I think it should be safe to ignore these for now, the problem is that some of the repositories, it has been some time since they were updated so there are some dependencies problems shown, these problems should be fixed when we release the new version of all the repositories! :smile:

IAmAMissingColon, 2019-11-08:
> Two questions:
>
> 1. Are you installing for development?
> 2. If django is working fine, I think it should be safe to ignore these for now, the problem is that some of the repositories, it has been some [...]

i) Installing For Development 
ii) Django Is Working Fine !

Okay ! :smile:

atb00ker, 2019-11-20:
Closing because it's a developement related issue that will be fixed after next releases. It's not affecting users to the best of my knowledge, please feel free to re-animate if required.

closed 2019-11-20 by atb00ker