openwisp/openwisp-controller #149: [bug] admin namespace required while using API

atb00ker, 2020-01-20:
I just hit the error:
Controller version: `0.4`
URL: `http://controller.atb00ker-test.org/api/device-location/<device-uuid>/`
Exception Type: `NoReverseMatch`
Exception Value: `'admin' is not a registered namespace`

I am not sure if `admin` namespace should be required to use API, this might be a bug, I may need to investigate but opening a bug report for now.

nemesisdesign, 2020-01-20:
@atb00ker maybe a full stack trace would help better.

atb00ker, 2020-01-20:
@nemesisdesign 
Traceback: https://paste.debian.net/plain/1126868
The error is happening because `admin` is not in the URLs, but I thought it's not required for controller API, that's why the issue is opened! :smile:

nemesisdesign, 2020-01-20:
@atb00ker look at:
```
 controller_1  |    File "/usr/local/lib/python3.7/site-packages/openwisp_utils/admin_theme/context_processor.py", line 8, in menu_items
 controller_1  |      menu = build_menu(request)
 controller_1  |    File "/usr/local/lib/python3.7/site-packages/openwisp_utils/admin_theme/context_processor.py", line 30, in build_menu
```

You should probably remove `admin_theme` in that container.

Looks like an issue in `docker-openwisp` rathern than `openwisp-controller.` If so, please transfer the issue to `docker-openwisp` if you can (or let me know and I'll do it).