openwisp/openwisp-controller #155: 500 error when adding user - default ansible install

nickberry17, 2020-02-06:
Just finished installing via [ansible](https://github.com/openwisp/ansible-openwisp2) on a fresh Ubuntu 18.04.3 droplet.  Getting **Server Error (500)** when trying to add a user.  I had to run the playbook twice to get it to finish cleanly, once before rebooting and once after.  Unsure what to do next.

nemesisdesign, 2020-02-06:
Can you check the logs in `/opt/openwisp2/logs` and report please?

nickberry17, 2020-02-06:
celery.log
```
 -------------- celery@telcomgmt v4.4.0 (cliffs)
--- ***** ----- 
-- ******* ---- Linux-4.15.0-66-generic-x86_64-with-Ubuntu-18.04-bionic 2020-02-06 20:05:29
- *** --- * --- 
- ** ---------- [config]
- ** ---------- .> app:         openwisp2:0x7f910bbd34e0
- ** ---------- .> transport:   redis://127.0.0.1:6379/3
- ** ---------- .> results:     disabled://
- *** --- * --- .> concurrency: 10 (eventlet)
-- ******* ---- .> task events: OFF (enable -E to monitor tasks in this worker)
--- ***** ----- 
 -------------- [queues]
                .> celery           exchange=celery(direct) key=celery
                

[tasks]
  . openwisp_controller.connection.tasks.update_config

[2020-02-06 20:05:35,642: CRITICAL/MainProcess] Unrecoverable error: VersionMismatch('Redis transport requires redis-py versions 3.2.0 or later. You have 2.10.6',)
Traceback (most recent call last):
  File "/opt/openwisp2/env/lib/python3.6/site-packages/kombu/transport/virtual/base.py", line 921, in create_channel
    return self._avail_channels.pop()
IndexError: pop from empty list

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/openwisp2/env/lib/python3.6/site-packages/celery/worker/worker.py", line 205, in start
    self.blueprint.start(self)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/celery/bootsteps.py", line 119, in start
    step.start(parent)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/celery/bootsteps.py", line 369, in start
    return self.obj.start()
  File "/opt/openwisp2/env/lib/python3.6/site-packages/celery/worker/consumer/consumer.py", line 318, in start
    blueprint.start(self)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/celery/bootsteps.py", line 119, in start
    step.start(parent)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/celery/worker/consumer/connection.py", line 23, in start
    c.connection = c.connect()
  File "/opt/openwisp2/env/lib/python3.6/site-packages/celery/worker/consumer/consumer.py", line 405, in connect
    conn = self.connection_for_read(heartbeat=self.amqheartbeat)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/celery/worker/consumer/consumer.py", line 412, in connection_for_read
    self.app.connection_for_read(heartbeat=heartbeat))
  File "/opt/openwisp2/env/lib/python3.6/site-packages/celery/worker/consumer/consumer.py", line 436, in ensure_connected
    callback=maybe_shutdown,
  File "/opt/openwisp2/env/lib/python3.6/site-packages/kombu/connection.py", line 430, in ensure_connection
    callback, timeout=timeout)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/kombu/utils/functional.py", line 344, in retry_over_time
    return fun(*args, **kwargs)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/kombu/connection.py", line 283, in connect
    return self.connection
  File "/opt/openwisp2/env/lib/python3.6/site-packages/kombu/connection.py", line 839, in connection
    self._connection = self._establish_connection()
  File "/opt/openwisp2/env/lib/python3.6/site-packages/kombu/connection.py", line 794, in _establish_connection
    conn = self.transport.establish_connection()
  File "/opt/openwisp2/env/lib/python3.6/site-packages/kombu/transport/virtual/base.py", line 941, in establish_connection
    self._avail_channels.append(self.create_channel(self))
  File "/opt/openwisp2/env/lib/python3.6/site-packages/kombu/transport/virtual/base.py", line 923, in create_channel
    channel = self.Channel(connection)
  File "/opt/openwisp2/env/lib/python3.6/site-packages/kombu/transport/redis.py", line 507, in __init__
    self.Client = self._get_client()
  File "/opt/openwisp2/env/lib/python3.6/site-packages/kombu/transport/redis.py", line 975, in _get_client
    'You have {0.__version__}'.format(redis))
kombu.exceptions.VersionMismatch: Redis transport requires redis-py versions 3.2.0 or later. You have 2.10.6
 
 -------------- celery@telcomgmt v4.4.0 (cliffs)
--- ***** ----- 
-- ******* ---- Linux-4.15.0-66-generic-x86_64-with-Ubuntu-18.04-bionic 2020-02-06 20:05:35
[... 118 lines cut ...]
spawned uWSGI worker 2 (pid: 14961, cores: 2)
[pid: 14961|app: 0|req: 1/1] 1.128.110.200 () {48 vars in 881 bytes} [Thu Feb  6 10:46:32 2020] GET / => generated 0 bytes in 640 msecs (HTTP/1.1 302) 5 headers in 156 bytes (1 switches on core 0)
[pid: 14961|app: 0|req: 2/2] 1.128.110.200 () {48 vars in 893 bytes} [Thu Feb  6 10:46:33 2020] GET /admin/ => generated 0 bytes in 46 msecs (HTTP/1.1 302) 9 headers in 386 bytes (1 switches on core 1)
[pid: 14961|app: 0|req: 3/3] 1.128.110.200 () {46 vars in 855 bytes} [Thu Feb  6 10:46:34 2020] GET /admin/login/?next=/admin/ => generated 2082 bytes in 179 msecs (HTTP/1.1 200) 8 headers in 439 bytes (1 switches on core 0)
[pid: 14961|app: 0|req: 4/4] 1.128.110.200 () {54 vars in 1105 bytes} [Thu Feb  6 10:46:37 2020] POST /admin/login/?next=/admin/ => generated 2243 bytes in 502 msecs (HTTP/1.1 200) 8 headers in 439 bytes (1 switches on core 1)
[pid: 14961|app: 0|req: 5/5] 1.128.110.200 () {54 vars in 1105 bytes} [Thu Feb  6 10:46:42 2020] POST /admin/login/?next=/admin/ => generated 0 bytes in 524 msecs (HTTP/1.1 302) 10 headers in 611 bytes (1 switches on core 0)
[pid: 14961|app: 0|req: 6/6] 1.128.110.200 () {48 vars in 942 bytes} [Thu Feb  6 10:46:43 2020] GET /admin/ => generated 9185 bytes in 95 msecs (HTTP/1.1 200) 7 headers in 263 bytes (1 switches on core 1)
[... cut ...]
[pid: 14959|app: 0|req: 1/11] 1.128.110.200 () {48 vars in 991 bytes} [Thu Feb  6 10:47:45 2020] GET /admin/openwisp_users/user/add/ => generated 14062 bytes in 1361 msecs (HTTP/1.1 200) 8 headers in 440 bytes (1 switches on core 0)
Thu Feb  6 10:48:34 2020 - *** HARAKIRI ON WORKER 2 (pid: 26245, try: 1) ***
Thu Feb  6 10:48:34 2020 - HARAKIRI !!! worker 2 status !!!
Thu Feb  6 10:48:34 2020 - HARAKIRI [core 0] 1.128.110.200 - POST /admin/openwisp_users/user/add/ since 1580986093
Thu Feb  6 10:48:34 2020 - HARAKIRI !!! end of worker 2 status !!!
DAMN ! worker 2 (pid: 26245) died, killed by signal 9 :( trying respawn ...
Respawned uWSGI worker 2 (new pid: 26538)
```