openwisp/openwisp-controller #157: OneToOneField to Device model class is missing Add/Edit buttons

okraits, 2020-02-12:
I have a model class which has a OneToOne relation to `openwisp_controller.config.models.Device` and to other models of mine:
```
    device = models.OneToOneField(Device, on_delete=models.CASCADE)
    dsl = models.OneToOneField(DslConnection, on_delete=models.CASCADE)
    sim = models.OneToOneField(SimCard, on_delete=models.CASCADE)
```
The issue is that I get Add and Edit buttons for all models but for the Device model:
[image]

okraits, 2020-02-23:
@nemesisdesign No idea regarding this?

nemesisdesign, 2020-02-25:
I don't think this is an issue with openwisp, rather it may be how the django admin works.

I suggest to set this model as an inline of `DeviceAdmin`.

nemesisdesign, 2020-02-25:
Closing this as I don't think it's a bug.

closed 2020-02-25 by nemesisdesign

okraits, 2020-02-25:
@nemesisdesign If this would be an issue with the django admin, then no OneToOneField would have Add and Edit buttons. But the other OneToOneFields have those buttons, only the OneToOneField using the `Device` model is missing those buttons.

If this is not a bug, is it intended behaviour? If yes, why? I would assume that OpenWISP models behave similar to upstream django models.

nemesisdesign, 2020-02-25:
@okraits The models of OpenWISP do not have any restriction on that side, I don't know why it is happening, but the bug report does not have enough information.

If you can find out more information about why you think this is a bug in OpenWISP we can re-open it and look for a fix. What I would do in this case is to go look into the django framework code which adds those buttons, drop an `import ipdb; ipdb.set_trace()` just before and find out what is causing the interested lines to not be reached.

At the moment this report does not contain enough information and I am putting effort to keep the issue list clean in order to help new contributors to get up to speed fast in contributing.

Let us know if you find out more!

okraits, 2020-02-27:
@nemesisdesign I found the cause of the issue. In the model form, the widget for the affected field is explicitly set because a js callback for the onchange event is set. This causes the RelatedFieldWidgetWrapper to not be applied. I will try to only update the widget attributes to fix the issue.
Thank you for your help!

nemesisdesign, 2020-02-27:
Welcome! Is the problem in one of the openwisp modules or in your application's code?

okraits, 2020-02-27:
In my application's code :grimacing:

nemesisdesign, 2020-02-27:
Thanks a lot for the update.

okraits, 2020-02-28:
For the record: a fine way to set/update an attribute on the widget for a relationship field in the admin is this (in the `ModelAdmin`:

```
    def formfield_for_dbfield(self, db_field, request, **kwargs):
        field = super().formfield_for_dbfield(db_field, request, **kwargs)
        if db_field.name == "device":
            field.widget.attrs.update({'onchange': "selectSim()"})
        return field
```