openwisp/openwisp-controller #158: Docker Redis error

avitus-pereira, 2020-02-24:
Whenever I try to run the project using Docker by following the commands given, I get the following error -- 
`[2020-02-24 04:24:42,532: ERROR/MainProcess] consumer: Cannot connect to redis://localhost:6379/1: Error 99 connecting to localhost:6379. Address not available..` 
It keeps trying to connect to redis but never connects.

atb00ker, 2020-02-27:
Try using docker-compose : https://github.com/openwisp/openwisp-controller/blob/master/docker-compose.yml

avitus-pereira, 2020-03-09:
> Try using docker-compose : https://github.com/openwisp/openwisp-controller/blob/master/docker-compose.yml

I tried this with a fresh install of docker. Cloned the repo and followed the steps for using it with docker as given in the read me. Now i get an error during docker-compose build. I will post that error in a separate issue. So cant confirm if the docker-compose file u mentioned works.

avitus-pereira, 2020-03-09:
> Try using docker-compose : https://github.com/openwisp/openwisp-controller/blob/master/docker-compose.yml

getting the a similar kind of error 

`controller_1  | [2020-03-09 17:31:21,787: ERROR/MainProcess] consumer: Cannot connect to redis://localhost:6379/1: Error 99 connecting to localhost:6379. Cannot assign requested address..
controller_1  | Trying again in 6.00 seconds... (3/100)`

The entire output of docker-compose is as below 

`➜ docker-compose up
Recreating openwisp-controller_redis_1 ... done
Recreating openwisp-controller_controller_1 ... done
Attaching to openwisp-controller_redis_1, openwisp-controller_controller_1
redis_1       | 1:C 09 Mar 2020 16:36:13.698 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
redis_1       | 1:C 09 Mar 2020 16:36:13.698 # Redis version=5.0.7, bits=64, commit=00000000, modified=0, pid=1, just started
redis_1       | 1:C 09 Mar 2020 16:36:13.698 # Warning: no config file specified, using the default config. In order to specify a config file use redis-server /path/to/redis.conf
redis_1       | 1:M 09 Mar 2020 16:36:13.702 * Running mode=standalone, port=6379.
redis_1       | 1:M 09 Mar 2020 16:36:13.702 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
redis_1       | 1:M 09 Mar 2020 16:36:13.702 # Server initialized
redis_1       | 1:M 09 Mar 2020 16:36:13.702 # WARNING you have Transparent Huge Pages (THP) support enabled in your kernel. This will create latency and memory usage issues with Redis. To fix this issue run the command 'echo never > /sys/kernel/mm/transparent_hugepage/enabled' as root, and add it to your /etc/rc.local in order to retain the setting after a reboot. Redis must be restarted after THP is disabled.
redis_1       | 1:M 09 Mar 2020 16:36:13.703 * DB loaded from disk: 0.000 seconds
redis_1       | 1:M 09 Mar 2020 16:36:13.703 * Ready to accept connections
controller_1  | Operations to perform:
controller_1  |   Apply all migrations: account, admin, auth, config, connection, contenttypes, geo, openwisp_users, pki, reversion, sessions, sites, socialaccount
controller_1  | Running migrations:
controller_1  |   Applying contenttypes.0001_initial... OK
controller_1  |   Applying contenttypes.0002_remove_content_type_name... OK
controller_1  |   Applying auth.0001_initial... OK
controller_1  |   Applying auth.0002_alter_permission_name_max_length... OK
controller_1  |   Applying auth.0003_alter_user_email_max_length... OK
controller_1  |   Applying auth.0004_alter_user_username_opts... OK
controller_1  |   Applying auth.0005_alter_user_last_login_null... OK
controller_1  |   Applying auth.0006_require_contenttypes_0002... OK
controller_1  |   Applying auth.0007_alter_validators_add_error_messages... OK
controller_1  |   Applying auth.0008_alter_user_username_max_length... OK
controller_1  |   Applying openwisp_users.0001_initial... OK
controller_1  |   Applying account.0001_initial... OK
controller_1  |   Applying account.0002_email_max_length... OK
controller_1  |   Applying admin.0001_initial... OK
controller_1  |   Applying admin.0002_logentry_remove_auto_add... OK
controller_1  |   Applying admin.0003_logentry_add_action_flag_choices... OK
controller_1  |   Applying auth.0009_alter_user_last_name_max_length... OK
controller_1  |   Applying auth.0010_alter_group_name_max_length... OK
controller_1  |   Applying auth.0011_update_proxy_permissions... OK
controller_1  |   Applying openwisp_users.0002_auto_20180508_2017... OK
controller_1  |   Applying openwisp_users.0003_default_organization... OK
controller_1  |   Applying openwisp_users.0004_default_groups... OK
controller_1  |   Applying pki.0001_initial... OK
controller_1  |   Applying config.0001_squashed_0002_config_settings_uuid... OK
controller_1  |   Applying config.0003_template_tags... OK
controller_1  |   Applying config.0004_add_device_model... OK
controller_1  |   Applying config.0005_populate_device... OK
controller_1  |   Applying config.0006_config_device_not_null... OK
controller_1  |   Applying config.0007_simplify_config... OK
controller_1  |   Applying config.0008_update_indexes... OK
controller_1  |   Applying config.0009_device_system... OK
controller_1  |   Applying config.0010_auto_20180106_1814... OK
controller_1  |   Applying config.0011_update_device_mac_address... OK
controller_1  |   Applying config.0012_auto_20180219_1501... OK
controller_1  |   Applying config.0013_last_ip_management_ip_and_status_applied... OK
controller_1  |   Applying config.0014_device_hardware_id... OK
controller_1  |   Applying config.0015_default_groups_permissions... OK
controller_1  |   Applying config.0016_default_organization_config_settings... OK
controller_1  |   Applying config.0017_template_name_organization_unique_together... OK
controller_1  |   Applying config.0018_config_context... OK
controller_1  |   Applying config.0019_organization_mac_add_hardware_id_name_unique_together... OK
controller_1  |   Applying config.0020_remove_config_organization... OK
controller_1  |   Applying config.0021_vpn_key... OK
controller_1  |   Applying config.0022_vpn_format_dh... OK
controller_1  |   Applying config.0023_update_context... OK
[... 22 lines cut ...]
controller_1  |   Applying socialaccount.0002_token_max_lengths... OK
controller_1  |   Applying socialaccount.0003_extra_data_default_dict... OK
[... cut ...]
controller_1  | [2020-03-09 17:36:31,072: ERROR/MainProcess] consumer: Cannot connect to redis://localhost:6379/1: Error 99 connecting to localhost:6379. Cannot assign requested address..
controller_1  | Trying again in 6.00 seconds... (3/100)
controller_1  |
controller_1  | [2020-03-09 17:36:37,096: ERROR/MainProcess] consumer: Cannot connect to redis://localhost:6379/1: Error 99 connecting to localhost:6379. Cannot assign requested address..
controller_1  | Trying again in 8.00 seconds... (4/100)
controller_1  |
^CGracefully stopping... (press Ctrl+C again to force)
Stopping openwisp-controller_controller_1   ... done
Stopping openwisp-controller_redis_1        ... done`

atb00ker, 2020-03-09:
We may need to point redis from `localhost:6379` to `redis:6379`!

avitus-pereira, 2020-03-10:
Yes that works. I hardcoded it in setting.py. Would be nice to pass it via .env or Dockerfile or something.

atb00ker, 2020-03-10:
If you can create a PR for fixing these issues with dockerfile and docker-compose file, I'd be most grateful! :smile:

avitus-pereira, 2020-03-10:
Sure will do that