openwisp/openwisp-controller #163: time incorrect

kapook007, 2020-03-08:
hi
 i have some problem openwisp  time incorrect .

my ubuntu already set timezone
how to fix ?

root@www:~# timedatectl
                      Local time: Sun 2020-03-08 22:37:09 +07
                  Universal time: Sun 2020-03-08 15:37:09 UTC
                        RTC time: Sun 2020-03-08 15:37:09
                       Time zone: Asia/Bangkok (+07, +0700)
       System clock synchronized: yes
systemd-timesyncd.service active: yes
                 RTC in local TZ: no
root@www:~#

okraits, 2020-03-08:
Please ask support questions on the mailing list :grinning:

closed 2020-03-08 by okraits

nemesisdesign, 2020-03-09:
Use one of: http://openwisp.org/support.html
This is a bug tracker, we need to focus on bugs and development related topics here.