openwisp/openwisp-controller #164: Docker Build Error

avitus-pereira, 2020-03-09:
Tried this on a fresh install of docker. After i run 
`docker-compose build`

I get the following error --

`ERROR: unsatisfiable constraints:
  so:libproj.so.19 (missing):
    required by: libspatialite-4.3.0a-r8[so:libproj.so.19]
ERROR: Service 'controller' failed to build: The command '/bin/sh -c apk add --no-cache             --update zlib-dev jpeg-dev libffi-dev gettext gcc openssl tzdata &&     apk add --no-cache             --repository http://dl-cdn.alpinelinux.org/alpine/edge/testing             --update geos-dev gdal-dev libspatialite &&     apk add --no-cache             --update             --virtual .build-deps postgresql-dev git build-base linux-headers openssl-dev' returned a non-zero code: 2`

atb00ker, 2020-03-09:
Oh I hate alpine for this reason, the remove packages, we'll need to move this image to Debian. :smile:

atb00ker, 2020-03-09:
I am not going to prioritize it because I doubt it affects a lot of people.
However, this may be a good starting point: 

```
RUN apt update && \
    apt install --yes --no-install-recommends zlib1g-dev libjpeg-dev \
                libcairo2 libpangocairo-1.0 gdal-bin gcc sqlite3 libsqlite3-dev openssl libssl-dev \
                gdal-bin libproj-dev libgeos-dev libspatialite-dev libsqlite3-mod-spatialits redis
```

avitus-pereira, 2020-03-09:
Small typo 
`libsqlite3-mod-spatialits --should be-- libsqlite3-mod-spatialite`

Thanks the the build works now

atb00ker, 2020-03-09:
P.S: We may need to point redis from `localhost:6379` to `redis:6379`, read here:  https://github.com/openwisp/openwisp-controller/issues/158#issuecomment-596640157

atb00ker, 2020-03-13:
I think this issue is not complete, the image is still the alpine one and according to the original question, the alpine image is not working, hence it still needs to be fixed.

avitus-pereira, 2020-03-15:
Yes. I changed it to debian and it was working. Didnt make any changes to the alpine version.

avitus-pereira, 2020-03-17:
I enabled the alpine community repo in the docker file which fixed the build issue. The build is working fine. Let me know if this is an acceptable fix.

`RUN echo -e "http://dl-cdn.alpinelinux.org/alpine/edge/community\nhttp://dl-cdn.alpinelinux.org/alpine/edge/main" >> /etc/apk/repositories`

There are some other errors which are more to do with versions which i will paste later.

atb00ker, 2020-03-18:
> Let me know if this is an acceptable fix.

I think it's better for us to move to Debian instead because alpine has been a maintenance hungry image, it keeps breaking!

dwang, 2020-10-05:
Looks like the Dockerfile was already changed to Debian in https://github.com/openwisp/openwisp-controller/commit/99b5366a3e32b212fc1e41362e508bafc3717e0e. I think this can be closed.

nemesisdesign, 2020-10-05:
Thanks everyone.

commit cbb6f7ba (2020-03-13): [docker] Added REDIS_URL to docker-compose.yml and settings.py #164  When running the project via docker, the controller tries to connect to redis using redis://localhost:6379 which should infact be redis://redis:6379. Added it as a environment variable named REDIS_URL in docker-compose.yml. This is read in settings.py and has a default value of redis://localhost:6397 if REDIS_URL is not passed to it. fixes #164
closed 2020-10-05 by nemesisdesign