openwisp/openwisp-controller #170: [models] device name must not be unique if hardware_id is enabled

okraits, 2020-03-17:
Right now, independent of the current settings, the device name is unique together with the organization.
But when `NETJSONCONFIG_HARDWARE_ID_ENABLED = True`, it should not be unique because then hardware_id is the unique field of the Device model.

nemesisdesign, 2020-03-17:
@okraits what is the use case for having different devices in the same organizations having the same name? It sounds like something confusing, why would you want to have different devices having identical name? Wouldn't it be better to always have different names anyway?

okraits, 2020-03-17:
The point is: if the hardware_id is enabled and thus the unique, identifying field of the device, the (host)name often doesn't matter and is not modified, thus the same default on all devices.
As a workaround, of course, the (host)name could be set to the same value as the hardware_id.

nemesisdesign, 2020-03-18:
In my experience, the people I work with want to have a human friendly name even if the hardware_id is enabled, because it makes it easy for them to manage the network.

Allowing duplicated names in the same organization does not make any sense to me.

I would rather let openwisp-config use the mac address for the hostname or, if the devices are created in the dashboard first, use the hardware ID as the name.

okraits, 2020-03-18:
I guess that usecases are different :grinning:

We have customers where the hostname and thus device name is set to a location specific value but we also have customers with > 10k devices which all have the same, default hostname because the hardware_id is the leading, unique and immutable identifier (which is also printed on the case of the device, as opposed to the hostname).

The hostname can be changed and there's no guarantee that it's really unique across all devices of an organization.

If you don't want to change the requirement of the device name to be unique or make that configurable, I might set the device name to the hardware_id or, if the customer is against that, keep a patch to remove the unique requirement.

nemesisdesign, 2020-03-22:
Ok let me know what you end up doing. Sounds more like a corner use case to me than the standard one.
What about just leaving the default hostname set by openwisp-config, that is, the mac address?

okraits, 2020-03-23:
That's not an option for us as the default hostname of our devices is the model, not the mac address.

nemesisdesign, 2020-03-26:
If you can change the firmware, you can add a uci-defaults script which sets the hostname to the mac-address format which is compatible with hostnames, like openwisp does, eg: '00-11-22-33-44-55'.

okraits, 2020-03-30:
Yes, but that changes the behaviour of our firmware which might not be what users want.

Is there any chance that we can make this (uniqueness of the device name) configurable?

A similar topic is the need to provide a mac address. If the hardware_id is enabled, it shouldn't be required to provide a mac address when creating a device on the controller. It would be sufficient if the mac would be set upon device registration. @nemesisdesign  What do you think?

nemesisdesign, 2020-03-30:
@okraits do you mean it's not possible to customize or recompile the firmware or what is the exact problem?

In some time we will have swappable models (thanks to GSoC, we already started warming up and implemented this in [openwisp-firmware-upgrader](https://github.com/openwisp/openwisp-firmware-upgrader/pull/9)) and that would allow you to customize the models as you wish. 

I don't see any use of putting effort in allowing devices to have all the same name, because if they all have the same name, it would be of no value and at this point it would make more sense to allow the field to be NULL and hide it completely from everywhere.

But this has to be designed and implemented with some care and sounds like lots of work to support a special case.

I think the options you have are:

A. add a [pre_registration_hook in openwisp-config](https://github.com/openwisp/openwisp-config/blob/master/openwisp-config/files/openwisp.agent#L135-L146) that sets a unique hostname to the devices

B. wait for swappable models

C. monkey patch to get the name to not be unique, handling the DB query to remove the unique constraint manually

okraits, 2020-04-01:
@nemesisdesign The firmware could be customized, of course. But the question is if customers are ok with the behaviour of the software changing. This is what we are evaluating right now.

What do you think regarding the mac address?

nemesisdesign, 2020-04-03:
@okraits keep in mind the mac address is assumed to be present by different parts. The context assumes is always present, the auto-cert generator as well. So changing this requires changing different parts, which implies risk of introducing bugs.

I have cases in which the hardware ID is based on the mac address, so I added some JS which automatically calculates the mac address.

I usually create a private django module where I keep these kind of customizations separate from the core code.

I'd look for a generalized solution only if it becomes common.

I suggest you to do the same.

You can achieve the same result now without the swappable model by monkey patching.
You can keep these monkey patching in the `ready()` method of the `AppConfig` of your custom app.

okraits, 2020-04-06:
@nemesisdesign The mac address will only be blank until the device registers at the controller, anyway. The requirement is only that a device can be created on the controller without having to enter the mac address.

Thank you for pointing me to `ready()` in `AppConfig` - I didn't know about that.

nemesisdesign, 2021-05-03:
**Update**:

See https://github.com/openwisp/openwisp-controller/issues/443 which once implemented will allow to configure the system to have non unique device names.

Another options is use `default_hostname` in openwisp-config, read it about there:
https://github.com/openwisp/openwisp-config#configuration-options

nemesisdesign, 2021-05-07:
Implemented in https://github.com/openwisp/openwisp-controller/pull/450, set `OPENWISP_CONTROLLER_DEVICE_NAME_UNIQUE = False` to disable the unique check.

closed 2021-05-07 by nemesisdesign