openwisp/openwisp-controller #176: [pki] Switch to UUIDs

nemesisdesign, 2020-03-22:
Switch PKI models to UUID in a backward compatible manner.

ManishShah120, 2021-01-25:
Hey @nemesisdesign, I would like to take up this issue.

Purhan, 2021-01-25:
@ManishShah120 you can go ahead :+1:

ManishShah120, 2021-01-25:
> @ManishShah120 you can go ahead

@Purhan I got assigned to a different Issue, for now, Maybe I'll start the work after completing that.

uzaxirr, 2023-01-22:
Hey, I would like to work on this, from what i've understood is that you wanna use a UUID as primary key for all the models in the `pki` django app. It would require changes in [openwisp-controller](https://github.com/openwisp/openwisp-controller/tree/master)/[openwisp_controller](https://github.com/openwisp/openwisp-controller/tree/master/openwisp_controller)/[pki](https://github.com/openwisp/openwisp-controller/tree/master/openwisp_controller/pki)/[base](https://github.com/openwisp/openwisp-controller/tree/master/openwisp_controller/pki/base)/models.py and [openwisp-controller](https://github.com/openwisp/openwisp-controller/tree/master)/[openwisp_controller](https://github.com/openwisp/openwisp-controller/tree/master/openwisp_controller)/[pki](https://github.com/openwisp/openwisp-controller/tree/master/openwisp_controller/pki)/models.py

Please correct me if i'm wrong

nemesisdesign, 2023-01-23:
@uzaxirr yes it's true but at the moment this is not an urgent change to make, if you want to contribute to OpenWISP and have a positive impact please join the dev chat https://gitter.im/openwisp/development and let's coordinate there so we can recommend issues that are going to have a positive impact in the short term, thanks!