openwisp/openwisp-controller #177: [device] Representation by hardware_id is not human-friendly

nemesisdesign, 2020-03-26:
Right now, if HARDWARE_ID is enabled, the device is represented using its hardware ID, even though in the cases I'm dealing with, the human readable name associated to the device is always preferred by users.

I would prefer to change the [__str__ method of Device to always use the name](https://github.com/openwisp/openwisp-controller/blob/87660dd0569742469f5a4f59f9da832192cd443d/openwisp_controller/config/models.py#L100), but from the discussion in #160 I understand that we may need to make this configurable.

@okraits what do you think?

okraits, 2020-03-26:
@nemesisdesign Indeed, if `__str__` would always use the device name, that would be a showstopper for some of our big customers.

nemesisdesign, 2020-04-02:
Made configurable in https://github.com/openwisp/django-netjsonconfig/pull/172, default behaviour maintained.

closed 2020-04-02 by nemesisdesign