openwisp/openwisp-controller #187: [tests] Avoid error logging to print noisy output during tests

nemesisdesign, 2020-04-11:
**Reserved for GCI**.

Depends on https://github.com/openwisp/openwisp-utils/issues/98.

Use the decorators mentioned in the issue linked above to silence the output of noisy tests.

2001asjad, 2020-11-01:
Hello, I would like to work on this issue.
But can you please guide me on which file to look on in which we have to make the change.

Thank You

2001asjad, 2020-11-02:
Hi,
I would like to work on this issue as I want to get started contributing to this organization, @nemesisdesign. But first can you please guide me on what files to look to work on this issue, I mean in what files and folders this issue would be found ,@atb00ker.
Thank You

atb00ker, 2020-11-02:
Run the `./runtests` command. 
Notice a lot of things are printed, ideally, only `.` would be printed for a passing testcase.
You need to checkout the `tests/` folder of the app, like the [config tests](https://github.com/openwisp/openwisp-controller/tree/master/openwisp_controller/config/tests).

Look at each test that is printing things that are not required and perform things like [redirecting stdout to `/dev/null`](https://stackoverflow.com/a/40417352/6410464) instead of the console.

2001asjad, 2020-11-02:
Thank You @atb00ker for your support. I will ask for the screenshot if I get stuck solving the issue😅, but do you know that why my migrations are taking too much time to run?

Thank You so much..

atb00ker, 2020-11-02:
It shouldn't take alot of time, reset and try again, if that doesn't help post your question on Gitter or on Mailing list to discuss further.

2001asjad, 2020-11-04:
It would be great if you attach the screenshots @atb00ker

2001asjad, 2021-01-08:
Hi @atb00ker, @nemesisdesign  I have started to work on this again. Whenever I run the tests, I get this : 
[image]

[image]
I wanted to ask that I have to use the decorators mentioned above only on the tests that have been highlighted red in the summary as slow tests , right?
Hope you'll help me in this .
Thank You

atb00ker, 2021-01-08:
No that slow test part is not relevant to this issue.
You only need to suppress the output in the second screenshot.
i.e it should not print things like `error: missing required parameter "key"`
When a test passes, it outputs a `.` and when it doesn't the output comes out after the test, those in-test-time outputs like "error: missing XYZ" does not help anyone, so we want to remove them!

2001asjad, 2021-01-08:
@atb00ker I ran `python runtests.py -v` to check which tests were noisy and in what file  but it gave me an error. Can you please help me in this as well. 
Thank you

2001asjad, 2021-01-09:
ok @atb00ker I'm on it.

PR #366 (2021-01-13): [controller-tests] Avoid error logging to print noisy output during t…
…ests #187

Identified and silenced all the noisy tests by using the @capture_any_output().

Fixes #187
commit 664f92e0 (2021-01-13): [tests] Avoid error logging to print noisy output during tests #187

Identified and silenced all the noisy tests by using the @capture_any_output().

Closes #187
commit fd20b516 (2021-04-09): [tests] Avoid error logging to print noisy output during tests #187

Identified and silenced all the noisy tests by using the @capture_any_output().

Closes #187
closed 2021-01-13 by nemesisdesign