openwisp/openwisp-controller #204: [controller/admin] Optimize queries of sortedm2m

nemesisdesign, 2020-06-09:
The sortedm2m widget generates too many unnecessary queries, we should find a way to avoid this:
https://github.com/jazzband/django-sortedm2m/issues/115

nemesisdesign, 2021-04-13:
I have thought about a possible solution for this.

By default, all the templates accessible to the user are included in the response.
On systems with many templates, this slows down the page a lot.

Now that we'll be loading the templates with JS only (#428), we could take this one step further.

We could edit the queryset of the sortedm2m admin field to not include any template:

```python
def formfield_for_manytomany(self, db_field, request, **kwargs):
        if db_field.name == 'templates':
            kwargs['queryset'] = Template.objects.none()
        return super().formfield_for_manytomany(db_field, request, **kwargs)
```

Then we'd have to edit the JS so that it handles the creation of HTML checkboxes (instead of showing what's already there).

Before implementing this fully, we should test if trying to save any option which is not present in the admin queryset works at all.
If it does, we should be able to do this.

DragnEmperor, 2025-04-30:
@nemesifier I have created a [PR](https://github.com/openwisp/openwisp-controller/pull/1010) for this. Let me know if the approach suits well with the requirement.

PR #1010 (2025-06-17): [change] Change relevant templates logic to facilitate changing organization +optimizations
## Checklist

- [x] I have read the [OpenWISP Contributing Guidelines](http://openwisp.io/docs/developer/contributing.html).
- [x] I have manually tested the changes proposed in this pull request.
- [x] I have written new test cases for new code and/or updated existing tests for changes to existing code.
- N/A I have updated the documentation.

## Reference to Existing Issue

Closes #204
Closes #1050

## Description of Changes

To remove queries generated by sortedm2m, 'formfield_for_manytomany' is over [... cut]
commit 2d7e5005 (2025-06-17): [change] Change relevant templates logic to facilitate changing organization +optimizations (#1010)

* [change] Reduce sortedm2m queries for templates #204

To remove queries generated by sortedm2m, 'formfield_for_manytomany'
is overriden such that on page loads, we can get templates using
'get_relevant_templates'. Added device/group id to identify selected
templates. Changes are made in JS to ensure templates are viewed and
changes are saved as per initial behaviour.
Ordering is not maintained.

Closes #204

* [fix] Fixed sorting issue in templates

Using the 'sort_value' of the 'through_mode [... cut]
commit c7567364 (2025-06-17): [change] Relevant templates: facilitate changing organization, optimize #204 #1050

Closes #204
Closes #1050
closed 2025-06-17 by nemesifier