openwisp/openwisp-controller #205: [bug] RelatedObjectDoesNotExist: Device has no config

nemesisdesign, 2020-06-16:
Full stack trace:

```python
RelatedObjectDoesNotExist: Device has no config.
  File "celery/app/trace.py", line 385, in trace_task
    R = retval = fun(*args, **kwargs)
  File "celery/app/trace.py", line 650, in __protected_call__
    return self.run(*args, **kwargs)
  File "openwisp_controller/connection/tasks.py", line 19, in update_config
    if device.config.status == 'applied':
  File "django/db/models/fields/related_descriptors.py", line 423, in __get__
    self.related.get_accessor_name()
```

It should not happen because this task is called only when a configuration is modified, but I guess it's happening because of the [database transaction issue](https://medium.com/hypertrack/dealing-with-database-transactions-in-django-celery-eac351d52f5f) (celery task is launched before the DB change is committed to the database).

If my supposition is correct, in order to replicate this, one should:

- run the dev server, run celery worker
- add a new device, without configuration, save
- add a configuration to the device, save

If the hypothesis is correct and the bug can be replicated, the solution is to use `transaction.on_commit` to send the config_modified signal only once the transaction has been committed to the DB.

Here's an example of similar usage in another module: https://github.com/openwisp/openwisp-firmware-upgrader/blob/4200399d3d71a0995896819f6259ebae9db9a64d/openwisp_firmware_upgrader/base/models.py#L86-L88

nemesisdesign, 2020-08-14:
quoting @NoumbissiValere 

I think it should be pretty safe to do two things:

- send this signal using `transacton.on_commit`: https://github.com/openwisp/openwisp-controller/blob/f3de849ce55e38ec1ebb6d6ecbec8dde190ccd71/openwisp_controller/config/base/config.py#L289-L300
- make this line resilient to failures: https://github.com/openwisp/openwisp-controller/blob/master/openwisp_controller/connection/tasks.py#L19 as @nepython did in https://github.com/openwisp/openwisp-monitoring/pull/192/files#diff-5b0ebdf698e7ab39e8c703b2a0d81b5bR35-R39 (writing the tests for it in a similar way)

nemesisdesign, 2020-08-14:
quoting @NoumbissiValere 

> Concerning openwisp/openwisp-controller#205, I have tested on my virtual openwrt device and didn't receive the same error you had.
> :disappointed: 

I think it should be pretty safe to do two things:

- send [this signal](https://github.com/openwisp/opeanwisp-controller/blob/f3de849ce55e38ec1ebb6d6ecbec8dde190ccd71/openwisp_controller/config/base/config.py#L289-L300) using `transacton.on_commit` (similar as we do in [this line of openwisp-firmware-upgrader](f3de849ce55e38ec1ebb6d6ecbec8dde190ccd71))
- make [this line](https://github.com/openwisp/openwisp-controller/blob/f3de849ce55e38ec1ebb6d6ecbec8dde190ccd71/openwisp_controller/connection/tasks.py#L19) resilient to failures:  as @nepython did in https://github.com/openwisp/openwisp-monitoring/pull/192/files#diff-5b0ebdf698e7ab39e8c703b2a0d81b5bR35-R39 (writing the tests for it in a similar way)

NoumbissiValere, 2020-08-14:
Ok @nemesisdesign I will try to implement this.

commit 10f4eb93 (2020-08-17): [bug] RelatedObjectDoesNotExist: Device has no config #205

Fixed the issue by ensuring the config_modified signal
ends when the db transaction is committed and also made
the update_config task resilient to failure.

Fixes #205
commit 399f963c (2020-08-17): [bug] RelatedObjectDoesNotExist: Device has no config #205

Fixed the issue by ensuring the config_modified signal
ends when the db transaction is committed and also made
the update_config task resilient to failure.

Fixes #205
commit bfe9b7f6 (2020-08-24): [bug] RelatedObjectDoesNotExist: Device has no config #205

Fixed the issue by ensuring the config_modified signal
ends when the db transaction is committed and also made
the update_config task resilient to failure.

Fixes #205
commit 05d5834b (2020-08-26): [bug] RelatedObjectDoesNotExist: Device has no config #205

Fixed the issue by ensuring the config_modified signal
ends when the db transaction is committed and also made
the update_config task resilient to failure.

Fixes #205
closed 2020-08-28 by nemesisdesign