openwisp/openwisp-controller #208: [vpn] Increase DH to 2048

nemesisdesign, 2020-06-24:
openssl now requires the DH to be minimum `2048` length, now we're using `1024`.

There's one problem though: this operation is slow, so I think we should mock `subprocess.check_output` in `Vpn.dhparam`.

There's also a DH saved in the test data, we should update that one to be of 2048 legnth.

Existing VPNs should be updated too (loop over all vpns, generate a new DH and save).

atb00ker, 2020-06-24:
> I think we should mock subprocess.check_output

Hey can you please explain this in more detail?
I initially thought, we could use celery and run this async but I am not sure if that is what you mean, any pointers about that would be helpful! :smile:

nemesisdesign, 2020-06-25:
Mock is fine, it's a bit slow but not so much and it's not an operation that is run often.

PR #210 (2020-07-07): [vpn] changed VPN length to 2048 #208 
Closes #208 

~~**Blocked by:** #190~~  
~~**Blocked by:** #219~~
commit dff8658c (2020-07-07): [change] Changed VPN DH length to 2048 #208

Closes #208
closed 2020-07-07 by nemesisdesign