openwisp/openwisp-controller #215: JSON validation error when dealing with OpenVPN

fabricat, 2020-06-30:
While editing a template in the openwisp-controller, after having switched to "Advanced mode (raw JSON)" i see these errors:
[image]
The error disappears when I add any alphabetical character at the beginning of an hostname (see the active line above).

nemesisdesign, 2020-07-07:
Update: netjsonconfig is working fine.

The problem lies with the JSON schema validator of the advanced mode.

Here's how to replicate this issue:

- add template
- click on "configuration menu", scroll down to NTP settings, click on the checkbox to enable it
- hit save and continue, the template should be saved normally
- now go to advanced mode

Expected result: advanced mode should work normally.

Actual result: JSON schema validation errors are  raised immediately.

**Suggestion**: maybe we can try to update the JS library used for the advanced mode or its jsonschema validator to see if the issue is fixed there. @NoumbissiValere since you put your hands in that part, maybe you can find a solution for this.

nemesisdesign, 2020-07-09:
Should be fixed, thank you @NoumbissiValere!

PR #232 (2020-07-09): [bug] JSON validation error when dealing with OpenVPN #215
Fixed the issue by updating the regular expression of the
hostname being used

Fixes #215
commit fea14e47 (2020-07-09): [fix] JSON validation error when dealing with OpenVPN #215

Fixed the issue by updating the regular expression of the
hostname being used

Fixes #215
closed 2020-07-09 by nemesisdesign