openwisp/openwisp-controller #217: [bug] Validation for name field broken

nemesisdesign, 2020-07-03:
How to replicate:

- go to the add device page
- use a name containing spaces (not an hostname)
- save

Expected result:

- validation error

Actual result:

- device is saved anyway

We need a regression test for this.

I would take advantage of this chance to add a `help_text` to the name field like the following:
`Will be used as hostname (must be a valid hostname).`

NoumbissiValere, 2020-07-05:
@nemesisdesign this issue was cause by the `absent` of  the `device_name_validator`  in the `name` field of `Device` model on the master branch. But this has been resolved here https://github.com/openwisp/openwisp-controller/blob/atb00ker-gsoc2020/openwisp_controller/config/base/device.py#L28.

So, I think we should close this issue and wait for `atb00ker-gsoc2020` branch to be merged to master or do you still want that submit a Fix for it?

nemesisdesign, 2020-07-05:
@NoumbissiValere no it's ok, thanks

nemesisdesign, 2020-07-06:
Confirm that was fixed in the latest master, thanks @NoumbissiValere @atb00ker :+1:

closed 2020-07-06 by nemesisdesign