openwisp/openwisp-controller #282: [bug] Notifications: Cannot read property '0' of null

nemesisdesign, 2020-09-17:
How to replicate:

- open add vpn server page
- look at the JS console

[image]

[image]

@pandafy can you help out fix this?

pniaps, 2020-09-17:
It happends in all 'Add' pages, but not when editing a object.

pandafy, 2020-09-18:
@pniaps Thanks for investigating it further. 

Earlier, I was under the presumption that Django uses a separate template for generating `add_view`. After looking into Django's code, I found out my assumption was wrong. Since, this problem lies in openwisp-notifications, I have created [an issue](https://github.com/openwisp/openwisp-notifications/issues/128) and [patch](https://github.com/openwisp/openwisp-notifications/pull/129) there. Using that patch, this error should not occur. I have tested it in both openwisp-controller and openwisp-monitoring.

nemesisdesign, 2020-09-18:
Cool, it's fixed in https://github.com/openwisp/openwisp-notifications/pull/129, but I think we should leave this open until we release a new bugfix version of openwisp-notifications and update https://github.com/openwisp/openwisp-controller/blob/master/requirements.txt#L11 accordingly, I also think this should be removed: https://github.com/openwisp/openwisp-controller/blob/master/install-dev.sh#L4-L5 so we can take advantage and do these changes when closing this issue.

nemesisdesign, 2020-10-18:
Related to https://github.com/openwisp/openwisp-controller/issues/308

PR #309 (2020-10-20): [chores] Updated openwisp_notifications to 0.2 #308
- Removed openwisp_notifications from `install-dev.sh`
  (openwisp_notification==0.2.1 is already released)
- Removed openwsip_users from `install-dev.sh`
  (openwisp_users==0.4.1 is already released)

Closes #308
Closes #282
commit 488a40f6 (2020-10-20): [chores] Updated openwisp_notifications to 0.2 #308

- Removed openwisp_notifications from `install-dev.sh`
  (openwisp_notification==0.2.1 is already released)
- Removed openwsip_users from `install-dev.sh`
  (openwisp_users==0.4.1 is already released)
- Added `rest` dependencies to openwisp_utils

Closes #308
Closes #282
closed 2020-10-20 by nemesisdesign