openwisp/openwisp-controller #287: [feature] Make Device verbose_name configurable

nemesisdesign, 2020-09-22:
In some cases, openwisp controller is used in specialized use cases, (Lora Gateways, WiFi hotspots, Routers), in these cases users prefer to use a more specialized word rather than the generic device.
We can easily accommodate this by making the `verbose_name` of `Device` configurable.

codesankalp, 2020-10-01:
@nemesisdesign I want to work on this issue.

NoumbissiValere, 2020-10-01:
You can go ahead @codesankalp

codesankalp, 2020-10-01:
Thannnks @NoumbissiValere

codesankalp, 2020-10-03:
@NoumbissiValere @nemesisdesign where should i add verbose_name for this issue?

pandafy, 2020-10-03:
@codesankalp  you will need to make verbose_name of Device model configurable. The place to add verbose_name is [Meta class of Device model](https://github.com/openwisp/openwisp-controller/blob/a85b9d6ecbdb8065634a66edd55b73f355310216/openwisp_controller/config/base/device.py#L79-L86). Please refer to [Django's documentation to learn more about verbose_name and other Meta options](https://github.com/openwisp/openwisp-controller/blob/a85b9d6ecbdb8065634a66edd55b73f355310216/openwisp_controller/config/base/device.py#L79-L86) if you are in doubt. 

You can take a look at implementation of other configurable settings like `HARDWARE_ID_OPTIONS` for reference. 

This is what I understand from the issue description, I hope @nemesisdesign meant the same, :smile:

codesankalp, 2020-10-03:
Thanks @pandafy for help :+1:

PR #296 (2020-10-10): [feature] Made Device verbose_name configurable #287
Added configurable verbose name field for Device name in the meta
class of AbstractDevice in device.py.

This value for this verbose_name is being imported from variable
OPENWISP_DEVICE_VERBOSE_NAME defined in settings.py.

Fixes #287
commit 6198d351 (2020-10-10): [feature] Made Device verbose_name configurable #287

Closes #287
commit 38828d2a (2020-10-10): [feature] Made Device verbose_name configurable #287

Closes #287
commit 53da00c0 (2020-10-10): [feature] Made Device verbose_name configurable #287 (#302)

Closes #287
closed 2020-10-10 by nemesisdesign