openwisp/openwisp-controller #293: [controller] Device name changed by a template is not updated in the device model instance

okraits, 2020-09-24:
We have a template to change the device hostname based on the location where it's in use:

```
{
    "system": [
        {
            "config_name": "system",
            "config_value": "system",
            "hostname": "TSL-{{ LOCATION_NAME }}"
        }
    ]
}
```
There's one issue with this, though:

The device name in the device model instance is not updated because:

* it is not sent to the device by `openwisp-config`, for example by `update_info`: https://github.com/openwisp/openwisp-config/blob/master/openwisp-config/files/openwisp.agent#L353
* it's not included `UPDATABLE_FIELDS`: https://github.com/openwisp/openwisp-controller/blob/64598a775d641d706da4ac22cf5ffbd7a3889792/openwisp_controller/config/controller/views.py#L129

I think it's a valid usecase to update the device hostname with a template and thus have this change reflected in the device model instance.

@nemesisdesign What do you think?

nemesisdesign, 2020-10-01:
@okraits the flow is unusual, you set the name of the device which is then modified by the device after it downloads the configuration specified in a template.

Where is `LOCATION_NAME` coming from? Is surely coming from somewhere isn't? Can't you simply script the name changing in another way that is internal to your application?

okraits, 2020-10-01:
> @okraits the flow is unusual, you set the name of the device on openwisp which is then modified by the device itself after it downloads the configuration specified in a template on OpenWISP. It's a bit strange.

The flow only uses core functionality of OpenWISP, I wouldn't call that unusual:

* The device registers itself at the controller
* the uci configuration on the device (here the hostname) is changed by a template which uses a context variable

IMO the device info on the controller should be updated if any device information present on the controller is outdated, be it the hostname, the firmware version or one of the IPs. Device info on the controller is of no use if it is outdated. The hostname can change if a device is migrated from one location to another one.

> Where is `LOCATION_NAME` coming from? Is surely coming from somewhere isn't?

It's a context variable in the device config on the controller.

> Can't you simply script the name changing in another way that is internal to your application?

Why would I want to do that? AFAIK configuration management should be the job of OpenWISP.

okraits, 2020-10-05:
@nemesisdesign How can we solve this? I think the implementation wouldn't take much effort - the client needs to send the hostname with the other infos and the controller needs to take that data and update the device object.

okraits, 2020-10-13:
@nemesisdesign I can't wait because I need this to be fixed, so I implemented a fix.

nemesisdesign, 2021-10-15:
> @nemesisdesign Did you have a chance to test or think about this?

I am still focused on delivering the next major release of OpenWISP.

code4me2all, 2026-10-22:
As a school, we also need this functionality. We have approximately 100 APs that should automatically register with OpenWISP after the firmware has been deployed and they are connected to the correct LAN.

LLDP is then used to identify the neighboring switch, including its name and port, which are set as the AP's new name. Ideally, this name should also be updated in OpenWISP to make it easier and faster to locate the APs.