openwisp/openwisp-controller #301: [JS] Upgrade jsoneditor library

nemesisdesign, 2020-10-08:
The JSON editor library which we've been using is not maintained anymore but there's a maintained fork: https://github.com/json-editor/json-editor.

It may be a good idea to migrate to the new library, porting our changes to it, so that at a leter moment we look for a way to introduce our changes without the need of changing the library.

@NoumbissiValere, since you put your hands on that part quite a few times, do you think you can help us with this?

NoumbissiValere, 2020-10-08:
@nemesisdesign I will look into it 😊

nemesisdesign, 2020-11-10:
Removed this from urgent priorities since we don't need it urgently. Moved it to the major release board to track long term goals.