openwisp/openwisp-controller #311: [JS] variable validation is broken

NoumbissiValere, 2020-10-26:
How to replicate: 

- In trying to add a new template, define a configuration variable with a valid value
- Use this variable in your configuration.
**Expected behaviors:**
- If the variable's value is valid according to the schema, the frontend (JS) validation should be ok.
- If variable's value is invalid according to the schema, the frontend validation should complain and the user shouldn't be allowed to save the template.
**noticed behaviors:**
- The frontend validation complains irrespective of if the variable's value is valid or not
- The template is saved with the complaining frontend validation.

This task will be consider done when the following conditions are met:

- [ ] validation errors should not be raised if the value in variables is correct
- [ ] validation errors should be raised if the value is bogus (or the variable does not exist or the variable contains an invalid value) and the saving operation must be blocked
- [ ] both previous cases have to be handled in normal mode and advanced mode

nemesisdesign, 2020-10-27:
Thanks for reporting @NoumbissiValere, please can you send a patch?

NoumbissiValere, 2020-10-27:
I am working on it 😄

commit 2e967e5f (2020-10-29): [fix] Fixed broken variablle validation #311

Fixes #311
commit 896a7f82 (2020-11-03): [fix] Fixed broken variablle validation #311

Fixes #311
commit c1c5ec9d (2020-11-03): [fix] Fixed broken variablle validation #311

Fixes #311
commit 33714195 (2020-11-04): [fix] Fixed broken variablle validation #311

Fixes #311
closed 2020-11-05 by nemesisdesign