openwisp/openwisp-controller #315: [bug] Schema validation for IP address is broken

NoumbissiValere, 2020-10-30:
- How to replicate:
try to create a template with an interface and set the IP address to an invalid value like `4..4.` and hit save. Schema validation should complain about the invalid IP address. without changing it, hit save again

- Expected behavior:
 irrespective of how many times you hit save with the invalid IP address, the form should not get save with an invalid IP address. meaning the schema validation should keep complaining. 

- Noticed behavior:
When you hit the save button at least twice with the invalid IP address, the form get save. 

We should look for a way to correct this.

dwang, 2020-11-06:
Also fixed by #313

nemesisdesign, 2020-11-06:
@NoumbissiValere do you confirm?
Thx @dwang

NoumbissiValere, 2020-11-09:
Yes @nemesisdesign #313 fixed it.

closed 2020-11-09 by nemesisdesign