openwisp/openwisp-controller #325: [deps] Update celery to version 5

nemesisdesign, 2020-11-19:
Depends on https://github.com/openwisp/openwisp-notifications/issues/155.

[Should 5 brings some breaking changes](https://docs.celeryproject.org/en/stable/whatsnew-5.0.html#whatsnew-5-0).

We should ensure the command to launch celery that is indicated in the README works and if not we need to update it.
Other than ensuring all tests pass, it's advised to do some manual testing running celery in the development environment, trigger the celery tasks and ensure they work.

2001asjad, 2021-02-10:
@nemesisdesign I would like to work on this issue.

pandafy, 2021-02-10:
Go ahead @2001asjad! There is a similar PR opened: https://github.com/openwisp/openwisp-notifications/pull/161. The conversation in that PR will be helpful for you to proceed with this issue.

PR #582 (2021-12-27): [deps] Upgraded celery~=5.1.2 #325
Closes #325
commit ed60c2e4 (2021-12-27): [deps] Upgraded celery~=5.1.2 #325

Closes #325
closed 2021-12-27 by nemesisdesign